MarlinFirmware/Marlin #12004: Problems with conventional jerk

AnHardt, 2018-10-04:
[image]
While searching for something completely different i realized, we have still problems with conventional jerk.

In the picture above you see a x-axis moving back and forth with 4 times jerk speed target. 
When we change the x-direction from low to high we see a nice beak, down to zero speed.
The high -> low transition is unrecognized.

The code looks as if we compare squares of speeds, where the change in sign is, logically, lost.
I guess we have to add a test if the (seemingly same speeds) have opposite singes. If so, and the speeds are above jerk/2, junction speed must be 0. Or something alike that. (I still don't see whats really going on - but dislike the result ;-( )

AnHardt, 2018-10-05:
```diff
@@ -2093,11 +2093,12 @@ bool Planner::_populate_block(block_t * const block, bool split_move,
       else delta_mm_i = ABS(delta_mm[i]);
     #else
     */
       const float delta_mm_i = ABS(delta_mm[i]);
     //#endif
-    const float cs = current_speed[i] = delta_mm_i * inverse_secs;
+    const float cs = delta_mm_i * inverse_secs;
+    current_speed[i] = delta_mm[i] * inverse_secs;
     #if ENABLED(DISTINCT_E_FACTORS)
       if (i == E_AXIS) i += extruder;
     #endif
     if (cs > max_feedrate_mm_s[i]) NOMORE(speed_factor, max_feedrate_mm_s[i] / cs);
   }
```
else later
```cpp
      #if ENABLED(JUNCTION_DEVIATION) && ENABLED(LIN_ADVANCE)
        LOOP_XYZ(axis)
      #else
        LOOP_XYZE(axis)
      #endif
      {
        // Limit an axis. We have to differentiate: coasting, reversal of an axis, full stop.
        float v_exit = previous_speed[axis] * smaller_speed_factor,
              v_entry = current_speed[axis];
        if (limited) {
          v_exit *= v_factor;
          v_entry *= v_factor;
        }

        // Calculate jerk depending on whether the axis is coasting in the same direction or reversing.
        const float jerk = (v_exit > v_entry)
            ? //                                  coasting             axis reversal
              ( (v_entry > 0 || v_exit < 0) ? (v_exit - v_entry) : MAX(v_exit, -v_entry) )
            : // v_exit <= v_entry                coasting             axis reversal
              ( (v_entry < 0 || v_exit > 0) ? (v_entry - v_exit) : MAX(-v_exit, v_entry) );

        if (jerk > max_jerk[axis]) {
          v_factor *= max_jerk[axis] / jerk;
          ++limited;
        }
      }
      if (limited) vmax_junction *= v_factor;
      // Now the transition velocity is known, which maximizes the shared exit / entry velocity while
      // respecting the jerk factors, it may be possible, that applying separate safe exit / entry velocities will achieve faster prints.
      const float vmax_junction_threshold = vmax_junction * 0.99f;
      if (previous_safe_speed > vmax_junction_threshold && safe_speed > vmax_junction_threshold)
        vmax_junction = safe_speed;
    }
...
  COPY(previous_speed, current_speed);
```
is useless because neither v_exit nor v_entry can be negative.
[image]
Untested on a real printer! Anyone?

AnHardt, 2018-10-05:
Caused by #11941
Ping @tcm0116

tcm0116, 2018-10-05:
I fail to see how #11941 is causing this issue. For non-mixing configurations, the planner code should be functionality equivalent. Can you provide more detail on what you suspect in #11941 is causing the issue?

GMagician, 2018-10-05:
@tmc0116 I think that `const float cs = ABS(current_speed[i] = delta_mm_i * inverse_secs)`
is different since `delta_mm_i` is always positive in your code and so `current_speed[i]` will be set to a always positive value, original may comes with `delta_mm_i` negative hence cs will be the same but not `current_speed[i]`

tcm0116, 2018-10-05:
The old code was:

```
const float cs = ABS((current_speed[i] = delta_mm[i] * inverse_secs));
```

`cs` is positive in both instances, but `current_speed[i]` could be negative in the old code.

I'll look at it and come up with a fix.

GMagician, 2018-10-05:
@tcm0116 that wat was I meant to say...sorry for my bad English

tcm0116, 2018-10-05:
@GMagician - we said the same thing. I just had to say it in a different way for me to understand it. Thanks for looking at it.

commit 49823a49 (2018-10-06): Fix broken jerk handling (#12019)

Fix #12004
closed 2018-10-06 by thinkyhead