MarlinFirmware/Marlin #12006: [Bugfix-2.0.x] M106 will not trigger fan, extruder fans will not start

ModMike, 2018-10-04:
### Description

Can not activate either fan on Rumba based board with verified fan pins 8 and 7.

### Steps to Reproduce

Send M106 command or start heating extruders

**Expected behavior:** [What you expect to happen]

Fans start

**Actual behavior:** [What actually happens]

Fans don't start

#### Additional Information

I am trying to finalize a Marlin installation on a Creatbot DX printer with a Rumba derived controller but I  hit a wall with the fans.  The controller has LEDs everywhere so it is easy to validate what MOFETs are being activated.

I am using the default example config and advanced config.

Here is what I have done to date:

1) Traced the pins directly to the 2560 using a multi-tester.  I am sure this is correct because I used the process to identify 3 of the heaters before I was clued in that it was a Rumba.  Those lined up perfectly with the Rumba, as did the steppers and end stops.  As expected, the fan pins were 2560 pins 17 and 16 which corresponds to Marlin PWM pins 8 and 7.

2) To make sure no other code was stepping on the pins, I commented out the fan pins 8 & 7 and  enabled debugging.  M43 reported that the pins were available.

3) I used an M42 to set the pin state to 255, still nothing.

At this point I am wondering if I misunderstood the #define E0_AUTO_FAN_PIN -1 statement.  I disabled the pins and set them to 8, or 7, and tested, still nothing.

The only thing I haven't done was flash a boot loader, is there ANY chance that could have an effect?

Desperate to wrap this up so all help very much appreciated!

ModMike, 2018-10-04:
It turns out that although the printer was working fine other than the fans, it turns out that the 12V voltage regulator that powered the fans had blown.  All fixed and working as expected.

Yeah I know, should have checked the hardware first!

closed 2018-10-04 by ModMike

thinkyhead, 2018-10-05:
Understandable. Hardware is usually the last thing we expect to have issues.