MarlinFirmware/Marlin #12007: Questions about pin output during Marlin startup.

JXPA, 2018-10-04:
Hello.
I am using Berd-Air to cool the PLA.
Yesterday, I replaced the air pump motor used with Berd-Air.
The new motor is a PWM-controlled brushless DC motor called A3650.

The replacement went well.
Motor is operating normally by connecting PWM wire to servo pin and assigning `FAN_PIN`.
However, after the power is turned on, the motor will run with full power until the Marlin startup process is completed.
The full power motor is very noisy.
Is there a way to specify the output state of a particular pin while Marlin's logo is displayed on the LCD?

thinkyhead, 2018-10-07:
Hmm. Try moving this line in `setup()` down…
```cpp
  servo_init();             // Initialize all servos, stow servo probe
```
…after…
```cpp
  #if ENABLED(SHOW_BOOTSCREEN)
    lcd_bootscreen();
  #endif
```

JXPA, 2018-10-07:
Thank you for your reply.
I tried it, but there was no change in the motor.
I do not know if it is related to this problem, but while flashing marlin, the motor is rotating at full power.

AnHardt, 2018-10-07:
A hardware reset sets all pins to a defined state. (For the AVRs that is 'input' with 'pullups low'.) If your motor is on in this state there is nothing we can do against that until the pin is initialized by Marlin. Especially we can't initialize it as log as the bootloader is running.
In that case only a hardware inverter does help. (That's why 'RAMPS FE V1' is called dangerous.)

JXPA, 2018-10-08:
Indeed I understand.
I will insert a suitable switch for the PWM wire.
Thank you.

closed 2018-10-08 by JXPA

MrAlvin, 2018-10-08:
@JXPA Like it has been said, then the default state (power on state / reset state) of the Atmega chips are input state.  So a pull-down resistor at the power-mosfet (or in your case a motor-on pin), will keep it from being activated. 

However, the Due default state of the pins is, with the internal pull-up resistor to be enabled. According to the datasheet, this pull-up resistor has a value between 50K and 150K, but is typically around 100K. 

So one way to make the default pin state different, is add an inverter between the Due pin and the motor controller on-pin. 

Another way is; to add a stronger pull-down resistor.  I have had good results with adding a 4K7 pull-down resistor to the Due pin(s), where I need the LOW signal state to be the default state on the pin.

A third way to change the default state of of your motor-on signal is, to add a switch between the Due pin, and the motor-on signal pin.  
And you need to check, that the motor controller already has a pull-down resistor on the motor-on pin off the motor controller. Otherwise you need to add such a pull-down resistor, even when adding a switch between the Due pin and the motor controller.

JXPA, 2018-10-09:
@MrAlvin - I appreciate your help.
It was solved by installing a 1k pull-down resistor.