MarlinFirmware/Marlin #12009: [Feature Request] Use PINDA 2 Probe with temperature compensation in Marlin

flyingscotsman12, 2018-10-04:
Would it be possible to include the temperature compensation from the thermistor in the PINDA 2 probe, like on the Mk3's? Prusa firmware already includes this, and it is awesome for making the inductive probes super accurate.

Ccleorina87, 2018-10-05:
I guess it's might not possible because there are so many types, size and torrance on inductive probe in market. Prusa was able to doing these because they only used one type of inductive probes and pursa make sure all probe are calibrate equally. But again it would be nice to have this function working on different probe other then Prusa PINDA probe.

boelle, 2018-10-05:
the function can only work if the probe has a thermistor inside it

that is what the pinda2 has and its also the reason for it having an extra wire

code wise i think it could be just a matter of taking the code from prusa firmware and putting it in marlin, prusa firmware is still just a stripped down marlin

StKob, 2018-10-21:
Im not that good at coding. Of course its not impossible but will take time.

thejerry1, 2019-02-03:
> Im not that good at coding. Of course its not impossible for me but will take time.

Did you have time to look on Prusa firmware? I failed to find any special properties for PINDA 2.
I got hands on PINDA 2 and I would like try it on Anet A8. If it's possible at all...

StKob, 2019-02-03:
Its not imposibble. PINDA 2 is just inductive switch with thermostat. It has 4 pins so its not plug&play. I was able to trace some parts of code for it but not enough time to analyse it.

AnHardt, 2019-02-04:
I'm tempted to disagree with about every statement here.
PINDA 2 has a thermistor - not a thermostat. Thermistor is a resistance based device for measuring temperatures. Thermostat is a device for holding a temperature static (constant). For that a measured temperature is compared with set temperature and the energy of a heater is regulated. PINDA 2 has no heater (what could solve some problems).
PINDA 2 is 'plug and play' when using/replacing the/a inductive sensor only.
Trying to copy Prusas code may not be the best option. Having a look into their github-bugtracker should cure you. Better would be to analyse their code to get their intention what to do and comparing that with what theory needs to be done. Finally judge if the target can be reached with the given methods, what limitations/compromises have to be done and/or what to improve.

StKob, 2019-02-04:
About thermostat, thats my mistake sorry. It is plug&play if you want to use it only as inductive probe. And I didnt want to make copy paste from their firmware. Thats usually a bad idea.

AnHardt, 2019-02-04:
Let's waste some thoughts on theory. Scope here is a Prusa i3 like machine only, to limit the size of this posting.
Target: Improve parts sticking on the bed by keeping the nozzle-bed distance at a ideal predetermined value. (Contrary to printing a raft, where the differing gap is filled by massive over-extruding of a first layer 'net' and ironing out that by the nozzle. The main problem here is separating the raft from the part.)
Contributing effects to the differing nozzles height over the bed: 
Mechanical: Bed not horizontal. Z-axis not vertical. Tilt x-axis. Smooth rods of any axis not parallel (rotation). Bed not flat.
Thermal: All parts size will differ with their temperature. Because thermal expansion of different materials is different, the parts relative size to each other differs with temperature.
Temporal: Every time you put energy into a 3D-printer system this is done **locally** by heating one of the heaters - producing a temperature gradient between the heater and its surrounding. The gradient is there as long as the heater delivers energy and/or the heaters environments temperature is not the same as the heaters. This equalization is done by conduction, convection and radiation. This takes time. For us that means  Even when our thermometer located near by the heater shows the target temperature the size of the related part is not its final size (hotend, probe, frame). 
For the bed (a not infinit thin plate) this means, as long there is a gradient between the top- and the bottom surface, the bottom surface is larger then the top surface, bending the plate into a bowl-shape where not fixed. If the fixing of the bed does not a allow for a bit of horizontal shift (bridge baring), at all but one point, the shape of the bed is completely undefined. It may snap from one shape into another. Discrete resistor heaters under the bed and compound beds (lower isolation, board material, board resistor material, electrical isolation, heat spreader, metal for the inductive probe, print surface) and whether the layers can move to each other or not further make the beds exact shape unpredictable. Realizing this means releasing; when printing the first layer we are partially coating the bed with a new (ideally fixed) layer - adding additional heat and isolation - changing the beds shape. The part-cooling fan, depending on how strong and where its cooling, changes the temperature gradient - and the beds shape.
In the PINDA the changing values of resistors, capacitors, frequencies have influence on to the probe-material distance the sensor triggers. Also the probes absolute length differs with temperature. (And its mount)

AnHardt, 2019-02-05:
With the Z/T [mm/°C] table we now can normalize every Z, at any place on the bed, when measured at the same time, measured temperature of the probe to a reference temperature, provided the temperature gradient in the probe system is low. The reference temperature can be any temperature in the measured interval. Selecting the highest or lowest makes most sense because the the sign of the correction is always the same then. (Test if the sign is correct. In the early versions Prusa had the wrong sign and 'corrrected' into the wrong direction.)
Now our Z measurements are independent from the probes temperature. With that we now could probe the bed at any bed-temperature without fearing wrong measurements because the probe changes temperature. For sure it is interesting how the beds shape alters with different bed temperatures. We even could make meshes at different bed temperatures. We even could try to interpolate between the probed meshes to get intermediate meshes - as long as the bed does not snap into an other shape at a temperature.
However. What we get (the shape of the bed) is not what we want (the difference between bed and nozzle)! We still have no information about the nozzles position. We do know the length of the nozzle-system (nozzle tip to x-carriage reference point) will vary with its temperature. We do know the temperature of the heater block, but have no clue about the temperature of the rest of the hotend system (heatsink, ...) because the heatbreak produces a wanted temperature gradient.

I don't know any way, neither automatically nor manually, how to make a table relating temperature to the length of the hotend - mostly because we have two temperatures - caused by the heatbreak.

**In the end we have not won much by measuring the probes temperature. If hotends length and probes length behave similar to each other when temperature changes we make it worse if we calculate out the probes errors but let the hotends in!.**

AnHardt, 2019-02-05:
I know this sequence was too long and too complicated for most.
Sentences grew and grew and some are messed up.

If you cant follow my thoughts - feel free to ask.
And, maybe i'm wrong.

TerRoshak, 2019-04-07:
TL;DR :
https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19890006009.pdf

GibsDev, 2019-06-19:
It looks like the process for detecting the thermal impact on the probe happens here: https://github.com/prusa3d/Prusa-Firmware/blob/5794debd0e85ad7e6216476a950092ba1d63230d/Firmware/Marlin_main.cpp#L4174-L4426 during Prusa's calibration process.

It looks like it does the following:
Wait for probe to be ambient temp
Probe bed
Turn on the heated bed and let heat from heated bed increase probe temp
Periodically probe the bed and see how many units/steps changed, store results in EEPROM table

I'm not sure it gets as complicated as the details @AnHardt goes into. From what I have seen the bed temperature is the only consideration when trying to make use of the PINDA probe thermistor.

cfunseth, 2019-07-25:
Would it be possible to complete this calibration using the heated bed's thermistor?  You could probe the bed "cold", then set the bed to various temperatures and use the existing Prusa logic outlined above.  

Since we're looking to correlate the offset to measured bed temperature, I'm not seeing a reason why the probe needs its own thermistor?  This would mean that any probe could be temperature compensated so long as we can read bed temperature.  Or am I missing something?

BrewChef, 2019-07-29:
> Would it be possible to complete this calibration just using the heated bed's thermistor and not the PINDA's? You could probe the bed "cold", then set the bed to various temperatures and use the existing Prusa logic outlined above.
> 
> Since we're looking to correlate the offset to measured bed temperature, I'm not seeing a reason why the probe needs its own thermistor? This would mean that any probe could be temperature compensated so long as we can read bed temperature. Or am I missing something?
> 
> Edit: As I think about this more, why don't we just probe the bed when it's at temperature? I think this is even suggested in the Marlin documentation. Do the Prusa printers probe before each print or just reference this temperature compensation and assume the bed hasn't changed?

It stopes the hot-end at 160c - heats the bed, probes, then resumes the hot-end heating.

GibsDev, 2019-08-22:
> Would it be possible to complete this calibration just using the heated bed's thermistor and not the PINDA's? You could probe the bed "cold", then set the bed to various temperatures and use the [...]

The problem we are trying to solve here is that the probe does not trigger at the same distance when it changes temperature. So when the probe is at room temperature, it will trigger a slightly shorter (but not insignificant) distance than when its at 50c. So if you can nail a first layer with your Z probe offset from a cold start, that's fine, but you run into problems when you try and print something immediately after. The probe will be warmer and it will trigger at a different (further) distance. This is very bad when it comes to homing the Z for the first layer because now your Z probe offset is not accurate. This causes the first layer to not be as close to the bed by a significant amount in my experience.

Technically if you were somehow using an endstop for Z homing, and only using the probe for gathering mesh data, this wouldn't really be an issue. This is not really possible for printers whose hotend's never leave the XY bounds of the build plate.

So to reiterate, the temperature of the probe itself is what causes the variation in measurement. Naturally it will change temperature due to changes in the environment around it. This includes the thermal changes produced by the hotend, the bed, cooling part fan, and room temperature. In terms of _accuracy_ of the probe, the probes temperature is all that matters (unless the inductive properties of the heated bed also significantly change with temperature). There is no guarantee that the temperature of the probe and the bed have any correlation, because there are so many other factors.

GibsDev, 2019-08-26:
I have successfully added the ability to read temperatures from the PINDA 2 probe in bugfix-2.0.x by using all of the temperature chamber code as reference. As this is my first time really diving into Marlin beyond configuration I'm not really sure where and how I should implement the rest.

Here is where I am currently working on it: https://github.com/GibsDev/Marlin/tree/pinda2

Assuming some variable that stores the current relative probe trigger distance in microsteps, where would I, as a dev familiar in marlins motion control framework (which I am not), apply that offset in code so that it does not overstep any existing Z calculations such as probe offset and bed leveling?

I would imagine the simplest solution would be to invisibly tack it on to the z probe offset, but is there actually a use case beyond that where trigger offset information is useful? Like for probing bed mesh points?

Canoivan, 2020-04-08:
Hello, did you manage to use pinda 2 in marlin?

closed 2020-04-08 by thinkyhead