MarlinFirmware/Marlin #12011: control ground via gcode

jorksx, 2018-10-04:
hello.
I have mks gen l board and I would like to know if I can control ground.
I have ender 3 so thats 24v system but I would like to connect rbg led strip to it without any mosfets or transistors.
microcontroller can handle the load so I just need to activate ground on 3 pins.
M42 p7 s120 
activates heater 2 that has my 12v led strip and s120 gives 12v
but my idea is to make something like this
with m42 p7 s120 I get 12v on heater 2 positive pin
I connect led strip positive to it
then I have pin 49 free (but its 5v when s255)
how can I make pin 49 (or any other digital pin "ground")
led strips are controlled with ground so if I activate p7 s120 and after that p49 to "ground" the led strip would light up. but p49 is 5v or 0v
I need to control ground pins somehow.
How?

gloomyandy, 2018-10-05:
Are you sure you only have 12V on that pin?If you are using a heater output with PWM to control the "output voltage" what you are measuring across the heater output pins is almost certainly an average voltage. In reality what you will probably have is a constant 24V output on one pin with a switched floating/ground output on the other (on most boards the heater and other switched outputs usually switch the ground output with a mosfet). It may not be a good idea to connect this to anything expect a real 12V output. Even worse the 12V you are seeing is because of the switched ground, if you connect the +ve pin  of your strip to this and then connect the -ve connection to some other pin that is held at a constant ground value then you will end up with the full 24V across those pins. Now you may get away with this, but I'm not sure it is a good idea unless you really know what you are doing.

Also are you sure that a normal I/O pin on the board can handle the current draw of the LED strip? Do you have a link to the strip you are planning to use?

jorksx, 2018-10-05:
yes, if arduino nano can handle 1 meter of ws2812b at full brightness, then my mks gen L can handle "led strip" that has only 3x 5050 smd leds ( so almost 3cm long )
and the 12v thingy I have done is good enough like this.
sure I can use npn transistors but all I have is mosfets that I cannot control with 5v or pnp transistors. And I cannot use pnp because they switch positive, and I really dont want to use relays.
Thats why I was thinking if I can switch ground 
sure I can use stepdown converter to get perfect 12v or close enough but I would still need to control the ground pins somehow

gloomyandy, 2018-10-05:
So exactly how do you intend to wire this strip up? Have you actually measured the voltage between full ground and whatever you intend to connect the strip +ve wire to? As I said above I think you will find it will end up being 24V not the 12V you are expecting. A link to the led strip you intend to use would help. What connections does it actually have.

jorksx, 2018-10-05:
from extruder 2 terminals, I get 12v. positive there to psu case is yes 24v.
I know.
They are reagular ebay led strip cut into 3cm piece.
4 wires, 1 positive and 3 ground
like I said I can use stepdown converter to get real 12v to led strip positive.
but I need to control the ground pins so I can switch different colors.

MrAlvin, 2018-10-09:
hmmm.... there are several ways that your reasoning of; what works => this must also work, that does not seem to make electrical sense. 

For instance; the ws2812b's are not powered by the data pin. They are powered by the + and - pins. The data pin of an Arduino nano is not sending any power to the ws2812b's. 
If you have connected + from your Arduino nano to + on your ws2812b's, then it is the Nano on-board regulator that supplies the power, not the Arduino pin. 
What kind of voltage did you supply your Nano with, when it was driving the ws2812b's?

Have you tested how many mA your 3x 5050 LED's use? 
Have you tested how many mA your 3x 5050 LED's use, at a 24V supply voltage? 

The complete load for an ATmega2560 chip must not exceed 200mA.  And this includes the 20-80mA that the MCU and the Arduino on-board LEDs are using. So it would be wise if you know how many mA your 5050 strips actually use. 

Now, if you want to use a 24V supply to power your 5050 led (12V) strips, and want to use your Arduino pin to "receive" the power to grund, then you will at the very least need to add a current limiting resistor in series with the LED strip.  
Lets say that one LED strip only allows 20mA to pass, and it uses 12V. Then this leaves 12V and 20mA to have to be handled by the current limiting resistor in series with the LED strip. So a 600 ohm resistor is needed. This resistor needs to be at least a 1/3 watt resistor. 
This is however a gambling way to implement this powering of a 3x 5050 LED strip, as we MUST make sure that there will NEVER be going more than 20-30mA into the Arduino pin, or it will be fried because the internal resistance of the internal Arduino pin mosfet will otherwise cause that the voltage over the Arduino input pin will become higher than 5V. And that WILL fry the pin. 

Now, the answer to your request of how to make the M42 command control a "ground" signal is to do:  
M42 P7 S0
As this would make the Arduino pin LOW, And this would turn the LED strip ON. 

To turn the LED strip OFF, you need to set the Arduino pin to the Z-state (= input). 
It will not be enough to set it to HIGH, as this will present 5V on the Arduino pin, and with a 24V power supply, this would still leave 24-5V = 19V over the LED strip and current limiting resistor, and would probably still make them appear to be ON. 
Your Arduino pin, MIGHT also see the 19V or 24V, and it would then be fried. But if we are lucky, then the input pin might actually try to sink the 20mA, and thus the voltage drop will happen over the LEDs and resistors, and leave the input pin at a potential of 0V. 

I have tested if a current limiting resistor can make an ATmega input pin become 12V tolerant, and this is possible. I did however test using a 10K current limiting resistor, and not only a 600 ohm current limiting resistor. 

I wonder why you think that you can not control a mosfet with a 5V signal?  
This is exactly what is being done by all RAMPS boards, and very, very likely also what is being done by your MKS board. 
N-channel mosfets, like to have 10V on the gate, but they will also allow some power to flow, if they only get 5V on the gate.  And some N-channel mosfets are even designed to be fully open with only 5V on the gate.  The IRL.... mosfets are such type of mosfets. The IRF..... type of mosfets like to have 10V on the gate, but when the load is only 3x 5050 LEDs, then the IRF..... type n-channel mosfets will also work.

After all of this, I would still say, that your safest bet is; to measure how many mA your 5050 LED strips actually use at 12V. 
And then calculate the right current limiting resistor, for use with a 24V supply. And then use a current limiting resistor, and a mosfet, to control each LED strip. 

Going directly into an arduino pin, is a gamble, and one day, something will go wrong, and then your Arduino pin is fried. And then what will you do? Replace the Arduino chip? Or will you need a completely different MKS board?

boelle, 2019-02-21:
@jorksx problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead