MarlinFirmware/Marlin #12012: [Bugfix-2.0.x] Serial -1 compile error

ModMike, 2018-10-05:
### Description

Compilation error when defining serial port as -1

### Steps to Reproduce

In config.h

```
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_RUMBA
```
Compiling produces the following error:

```
Compiling .pioenvs/megaatmega2560/src/src/HAL/HAL_AVR/persistent_store_eeprom.cpp.o
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/HAL.h:45:0,
from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_AVR/MarlinSerial.cpp:38:
Marlin/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/MarlinSerial.h:69:43: error: pasting "UDR" and "-" does not give a valid preprocessing token
#define M_UDRx             SERIAL_REGNAME(UDR,SERIAL_PORT,)

```
#### Additional Information

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2448566/Archive.zip)

Phr3d13, 2018-10-05:
What arduino ide version?

ModMike, 2018-10-05:
VS Code Platform IO

ejtagle, 2018-10-05:
Serial port should not be negative (except on some strange configuration, as for example DUE)
On AVR SERIAL_PORT should be 0 for the bootloader serial port

ModMike, 2018-10-05:
It used to compile fine and that option is referenced right in Marlins config file.

I am trying to configure a touch panel that acts like a host on a Rumba based board.

 From what I’ve read here, the config needs to be -1 for first serial port and 0 for second.  0 and 1 does not work.

On my Azteeg X5 Mini Wifi, that was the only config that allowed me to have a USB and Web UI host that was built in to the board.

If there is another way to do this I am all ears.

gloomyandy, 2018-10-05:
As above the -1 option only really applies to boards like the MKS SBase or DUE that have a USB controller built into the chip. I don't think there are any AVR boards that have this. The issue you referenced was for an SBase board (which does have the onboard USB), not for a Rambo (which as far as I can tell does not). On most (all?) AVR based boards the "USB connection" goes via an external chip that connects to a hardware USART on the AVR (and so you reference them as 0, 1, etc not -1). I think the Azteeg X5 makes use of a Smoothiboard style control board (which in effect is the same as the MKS Sbase), not an AVR based board.

ModMike, 2018-10-05:
That explains it.  Not sure what the Rumba uses but they have an Atmega16u2 to handle USB (I think).

ejtagle, 2018-10-07:
@ModMike : You must find out the serial ports that are actually used by your hardware.
For AVR, usually hardware serial port 0 is connected to a Atmega 16U2 that acts as a USB to serial bridge - So that should be the definition of ```SERIAL_PORT```
Now, you must find the other serial port number your panel is connected to. Maybe it is 1, maybe it is 2, or perhaps 3.
Just define SERIAL_PORT_2 as the appropiate number.

Support for multiple serial ports on AVR requires the latest bugfix-2.0.x  as that configuration was not supported previously

ModMike, 2018-10-10:
@ejtagle Thank you!  Sorry for the late reply, computer blew up.

closed 2018-10-12 by ModMike