MarlinFirmware/Marlin #12015: [bugfix-2.0.x] Error:EEPROM datasize error when enabled 2 servos (dondolo + bltouch)

alexxy, 2018-10-05:
Hi!

I get this error with recent bugfix-2.0.x branch. Configs attached
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2451053/config.zip)

```
Compiled: Oct  5 2018
Free Memory: 2724  PlannerBufferBytes: 1488
Error:EEPROM datasize error.
ttyUSB0 PC> Error:EEPROM datasize error.

Index: 607 Size: 615
Hardcoded Default Settings Loaded

```

alexxy, 2018-10-05:
yes. doesnt work for me. actualy it wasfirst thing that i tryed.

blairq, 2018-10-05:
im getting it too. also the machine started skipping steps and so... already did a m502 and m500. changed z_offset (i use a touch sensor) , saved the changes but they didnt go to the eeprom. 
flashed 2 times. same error.

GMagician, 2018-10-05:
@alexxy @blairq have you tried with DEBUG_EEPROM_READWRITE and check log

blairq, 2018-10-05:
no, its related to https://github.com/MarlinFirmware/Marlin/commit/c40ea1f0fd28f421bad4f25b660e2316e4c5f2e6

i changed back #define EEPROM_OFFSET 0 to the original #define EEPROM_OFFSET 100 and everythings works fine again.

mrambossek, 2018-10-06:
i can confirm that there seems to be some kind of issue. after upgrading last night, i keep getting
```
Error:EEPROM CRC mismatch - (stored) XXXXX != YYYYY (calculated)!
echo:Hardcoded Default Settings Loaded
```
i changed EEPROM_OFFSET back to 100 as suggested by https://github.com/MarlinFirmware/Marlin/issues/12015#issuecomment-427486004 and now it seems to work again. so appearently those 100 bytes ARE somehow used? :)

alexxy, 2018-10-06:
I tryed with DEBUG_EEPROM_READWRITE

```
SENDING:M500
Error:Field lcd_preheat_hotend_temp mismatch.
Error:Field lcd_preheat_hotend_temp mismatch.
```

alexxy, 2018-10-06:
Actualy error is the same as for #11883 and #11757

GMagician, 2018-10-06:
@thinkyhead in servo.h there is something not correct I think `extern uint16_t servo_angles[NUM_SERVOS][2];`. Not related with this but may lead to memory corruption since I think it should use `NUM_SERVO_PLUGS`

GMagician, 2018-10-06:
@thinkyhead sorry to bother but shouldn't I see `eeprom_index` changed every read/write? I see it is incremented only in `EEPROM_SKIP`

alexxy, 2018-10-06:
I have NUM_SERVOS  = 2  in Configuration.h 
But may be problem is that i use BLTOUCH? and enabled editable servo angles?

alexxy, 2018-10-06:
Disableing EDITABLE_SERVO_ANGLES solves issue (so seems like its a root of problem)

GMagician, 2018-10-06:
@alexxy not sure about problem but I think that this mixing of `NUM_SERVOS` and `NUM_SERVO_PLUGS` on `servo_angles` may lead to some problem...

GMagician, 2018-10-06:
@alexxy that confirm me that such mix is the issue...

alexxy, 2018-10-06:
NUM_SERVO_PLUGS not defined in Marlin/src/module/servo.h

GMagician, 2018-10-06:
If it will fix I'll post a PR

GMagician, 2018-10-06:
OT @alexxy your dondolo works fine? I decided to remove it because of too much problem with filament retraction when not in use

alexxy, 2018-10-06:
@GMagician it works. but its hard to calibrate offsets

GMagician, 2018-10-06:
@al

> 
> 
> @GMagician it works. but its hard to calibrate offsets

Dondolo? or fix? If dondolo is no a problem of offsets. The problem is that the unused filament is not kept in position when head moves. When you come back to such filament you miss material (you have to create a tower but then no benefit from having two nozzles)

alexxy, 2018-10-06:
Dondolo works.

cjsoong, 2018-10-06:
@alexxy 
If your problem has not been resolved, you can try my modified version.
No EEprom data size error & M281 can set the servo angle.
https://drive.google.com/file/d/1dixj-e1lBUvuxazjgOgqVz0tbXnl2zKI/view?usp=sharing

closed 2018-10-06 by thinkyhead

monterman, 2018-10-24:
@alexxy could you help me with my issue.   see here please https://github.com/teemuatlut/Marlin/issues/27

thank you.

commit e8d67762 (2018-10-06): Use NUM_SERVOS in EEPROM code

Fix #12015