MarlinFirmware/Marlin #12018: [bugfix-2.0.x] Error

Ccleorina87, 2018-10-06:
Error while compile latest Marlin 2.0.x

*** [.pioenvs\megaatmega2560\lib195\libU8glib-HAL_ID1932.a] Error 1

Sorry i have no idea what these error are...

Ludy87, 2018-10-06:
Your error Message is very small, post the details before this lines.

Ludy87, 2018-10-06:
Try "PlatformIO: Clean"

Ccleorina87, 2018-10-06:
> Try "PlatformIO: Clean"

Not working... i might reinstall vscode again... if it working after that

Ccleorina87, 2018-10-06:
> Try "PlatformIO: Clean"

Already try it... still the same... reinstall Vscode and platformio still the same... try older backup of Marlin 2.0.x 3 days ago working fine... im lost now... on latest one cant...

Ccleorina87, 2018-10-06:
> Try "PlatformIO: Clean"

funny things is... disable My Avast Antivirus and its working... now this is wired...

Ccleorina87, 2018-10-06:
> Try "PlatformIO: Clean"

i found it if i Enable these #define SDCARD_SORT_ALPHA the error comeback

thinkyhead, 2018-10-06:
You didn't fill out the template and provide all requested information, so we don't have enough information to help. Please ZIP your `Configuration.h` and `Configuration_adv.h` files and attach them to your next post.

Ccleorina87, 2018-10-06:
> You didn't fill out the template and provide all requested items, so we don't have enough information to help. Please ZIP your `Configuration.h` and `Configuration_adv.h` files and attach them to your next post.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2453502/Marlin.zip)

Ccleorina87, 2018-10-07:
i jump the gun saying the SDCARD_SORT_ALPHA... it's so random error... i try compile SDCARD_SORT_ALPHA disable again if fail to compile... enable also error... im lost now...

Ccleorina87, 2018-10-07:
[image]
[image]

I always get these error... sometime it successful... if i try build again get either these 2 error... i already reinstall VSCode and platformIO... i'm lost...

Ccleorina87, 2018-10-09:
> I just did a build of the current `bugfix-2.0.x` branch in PlatformIO (VSCode) using your configs (plus random `_Bootscreen.h` and `_Statusscreen.h`) and the build was successful.

Ok already fix... i close these issue... All run perfectly

closed 2018-10-09 by Ccleorina87