MarlinFirmware/Marlin #12020: Dual-Z-Axis Homing: Only perfect if hitting Z2_MIN_ENDSTOP first

MarphXL, 2018-10-06:
### Description

I Have MKS Gen V1.4 with TMC2208 on all stepper. I Use Z-Dual-Axis with two Endstops (Z-MIN and Z2-Min).
If I do a Z-homing both Z-axis going down.
If it hits the Z2_MIN_ENDSTOP first, the Z2-AXIS stops and the Z-AXIS going down till it hits the Z_MIN_ENDSTOP. Perfect! Both sides are leveled.

But if the Printer hits the Z-Min-Endstop first than he stops homing. The Z2-Min-Endstop wasn't hit.

It's reproducable. Everytime!

Could you help and give me a tip what I have misconfigured or missunderstanded?

configuration_adv.h
`#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  #define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _YMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif`

configuration.h
`// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_Z2MIN_PLUG // 20181006
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG  // ***
//#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS // ***
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN  // ***
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.`

`
//
// Limit Switches
//
#define X_MIN_PIN           3
#ifndef X_MAX_PIN
  #define X_MAX_PIN         -1   // D2 ist für serial /USB-Serial
#endif
#define Y_MIN_PIN          14
#define Y_MAX_PIN          15
//#define Z2_MIN_PIN         15   // wird in configuration_adv.h mit Y_Max definiert
#define Z_MIN_PIN          18
#define Z_MAX_PIN          19`

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2453168/Marlin.zip)

thinkyhead, 2018-10-07:
> #define USE_Z2MIN_PLUG // 20181006

There is no such thing.

We'll look at the dual-Z-endstop homing and see if there's anything obvious going on.

Have you tried `bugfix-2.0.x` to see if it exhibits the same issue?

thinkyhead, 2018-10-07:
Try this branch and see if it works any better for you: https://github.com/thinkyhead/Marlin/archive/bf1_old_dual_homing.zip

MarphXL, 2018-10-07:
Thanks but it doesn't help.

Hre are my pins from M43, if this help:

Recv: PIN:   0   Port: E0        RXD                         protected
Recv: PIN:   1   Port: E1        TXD                         protected
Recv: PIN:   2   Port: E4        <unused/unknown>            Input  = 1    TIMER3B   PWM:     0    WGM: 1    COM3B: 0    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
Recv: PIN:   3   Port: E5        X_MIN_PIN                   protected
Recv: PIN:   4   Port: G5        Y_SERIAL_RX_PIN             Input  = 1    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled
Recv: PIN:   5   Port: E3        Y_SERIAL_TX_PIN             Output = 1    TIMER3A   PWM:     0    WGM: 1    COM3A: 0    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
Recv: PIN:   6   Port: H3        X_SERIAL_RX_PIN             Input  = 1    TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
Recv: PIN:   7   Port: H4        CONTROLLER_FAN_PIN          protected
Recv: PIN:   8   Port: H5        E0_AUTO_FAN_PIN             protected
Recv: .                          RAMPS_D8_PIN                protected
Recv: PIN:   9   Port: H6        FAN_PIN                     protected
Recv: .                          RAMPS_D9_PIN                protected
Recv: PIN:  10   Port: B4        HEATER_0_PIN                protected
Recv: .                          RAMPS_D10_PIN               protected
Recv: PIN:  11   Port: B5        X_SERIAL_TX_PIN             Output = 1    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
Recv: PIN:  12   Port: B6        <unused/unknown>            Input  = 0    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
Recv: PIN:  13   Port: B7        LED_PIN                     Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
Recv:  .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
Recv: PIN:  14   Port: J1        Y_MIN_PIN                   protected
Recv: PIN:  15   Port: J0        Y_MAX_PIN                   protected
Recv: PIN:  16   Port: H1        LCD_PINS_RS                 Output = 0
Recv: PIN:  17   Port: H0        LCD_PINS_ENABLE             Output = 0
Recv: PIN:  18   Port: D3        Z_MIN_PIN                   protected
Recv: PIN:  19   Port: D2        <unused/unknown>            Input  = 1
Recv: PIN:  20   Port: D1        <unused/unknown>            Input  = 1
Recv: PIN:  21   Port: D0        <unused/unknown>            Input  = 1
Recv: PIN:  22   Port: A0        <unused/unknown>            Input  = 0
Recv: PIN:  23   Port: A1        LCD_PINS_D4                 Output = 1
Recv: PIN:  24   Port: A2        E0_ENABLE_PIN               protected
Recv: PIN:  25   Port: A3        LCD_PINS_D5                 Input  = 1
Recv: PIN:  26   Port: A4        E0_STEP_PIN                 protected
Recv: PIN:  27   Port: A5        LCD_PINS_D6                 Input  = 1
Recv: PIN:  28   Port: A6        E0_DIR_PIN                  protected
Recv: PIN:  29   Port: A7        LCD_PINS_D7                 Input  = 0
Recv: PIN:  30   Port: C7        Z2_ENABLE_PIN               protected
Recv: PIN:  31   Port: C6        BTN_EN1                     Input  = 1
Recv: PIN:  32   Port: C5        <unused/unknown>            Input  = 0
Recv: PIN:  33   Port: C4        BTN_EN2                     Input  = 1
Recv: PIN:  34   Port: C3        Z2_DIR_PIN                  protected
Recv: PIN:  35   Port: C2        BTN_ENC                     Input  = 1
Recv: PIN:  36   Port: C1        Z2_STEP_PIN                 protected
Recv: PIN:  37   Port: C0        BEEPER_PIN                  Output = 0
Recv: PIN:  38   Port: D7        X_ENABLE_PIN                protected
Recv: PIN:  39   Port: G2        <unused/unknown>            Input  = 0
Recv: PIN:  40   Port: G1        E0_SERIAL_TX_PIN            Output = 1
Recv: PIN:  41   Port: G0        KILL_PIN                    Input  = 1
Recv: PIN:  42   Port: L7        SD_DETECT_PIN               Input  = 1
Recv: .                          Z2_SERIAL_RX_PIN            Input  = 1
Recv: PIN:  43   Port: L6        <unused/unknown>            Input  = 0
Recv: PIN:  44   Port: L5        HEATER_BED_PIN              protected
Recv: PIN:  45   Port: L4        <unused/unknown>            Input  = 0    TIMER5B   PWM:     0    WGM: 1    COM5B: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0
Recv: PIN:  46   Port: L3        Z_STEP_PIN                  protected
Recv: PIN:  47   Port: L2        <unused/unknown>            Input  = 0
Recv: PIN:  48   Port: L1        Z_DIR_PIN                   protected
Recv: PIN:  49   Port: L0        <unused/unknown>            Input  = 1
Recv: PIN:  50   Port: B3        AVR_MISO_PIN                Input  = 1
Recv: .                          MISO_PIN                    Input  = 1
Recv: PIN:  51   Port: B2        AVR_MOSI_PIN                Input  = 0
Recv: .                          MOSI_PIN                    Input  = 0
Recv: PIN:  52   Port: B1        AVR_SCK_PIN                 Input  = 1
Recv: .                          SCK_PIN                     Input  = 1
[... cut ...]
Recv: PIN:  63   Port: K1 (A 9)  E0_SERIAL_RX_PIN            Input  = 1
Recv: PIN:  64   Port: K2 (A10)  Z_SERIAL_RX_PIN             Input  = 1
Recv: PIN:  65   Port: K3 (A11)  Z2_SERIAL_TX_PIN            Output = 1
Recv: PIN:  66   Port: K4 (A12)  <unused/unknown>            Output = 0
Recv: PIN:  67   Port: K5 (A13)  TEMP_0_PIN                  protected
Recv: PIN:  68   Port: K6 (A14)  TEMP_BED_PIN                protected
Recv: PIN:  69   Port: K7 (A15)  TEMP_1_PIN                  Analog in =  1023

MarphXL, 2018-10-07:
Waaaah... 
Same problem with bugfix 2.0. First I have had the problem with missing TMCStepper.h but I found the solution at Issue 11992.
Could it be the stepper-library that makes this problem?

MarphXL, 2018-10-07:
My last try was deleting alle stepper-libraries from arduino. Install latest tmcstepper-lib and compile again but it doesn't change the problem.
Did you have another tip?

MarphXL, 2018-10-07:
Nobody could help?

MarphXL, 2018-10-08:
patience. yes, sorry about that.
I have tested endstop noise filter but it change nothing. 
all my endstops are imverted. could this be a problem?

GMagician, 2018-10-08:
I'm still using dual Z driver and dual z endstop (but v2.0) and it works with no issues. Have you checked with M119 if endstops work correctly?

MarphXL, 2018-10-08:
yes, I tested M119. 
Did you have inverted endstops?

GMagician, 2018-10-08:
Yes, inverted, only difference I use _ZMAX_ instead of _YMAX_ as second input and I have interrupts enabled

thinkyhead, 2018-10-08:
Inverted (N-O) switches are more likely to have issues if they get shorted together, whereas N-C switches are caught earlier and are also immune to induction. You can use `PINS_DEBUGGING` and `M43` to check each endstop pin to make sure that when one endstop is triggered, the other one isn’t. (`M119` only shows one endstop state per axis, AFAIK.)

MarphXL, 2018-10-08:
thanks. could you paste your m43 and m119 so I can compare.

GMagician, 2018-10-08:
My M119 with no endstop pressed:
 22:33:44.065 : x_min: open
 22:33:44.065 : y_min: open
 22:33:44.065 : z_min: open
 22:33:44.065 : z2_min: open

My M119 with no Z1 endstop pressed:
 22:35:33.769 : x_min: open
 22:35:33.769 : y_min: open
 22:35:33.769 : z_min: TRIGGERED
 22:35:33.769 : z2_min: open

My M119 with no Z2 endstop pressed:
 22:36:21.798 : x_min: open
 22:36:21.798 : y_min: open
 22:36:21.798 : z_min: open
 22:36:21.798 : z2_min: TRIGGERED

GMagician, 2018-10-08:
@thinkyhead 

> M119 only shows one endstop state per axis, AFAIK.)

mine report both

@MarphXL sorry but M43 is not enabled in my firmware

thinkyhead, 2018-10-08:
> mine report both

Oh good. That makes it easier for testing.

MarphXL, 2018-10-11:
I checked all twice. My M119 is exact like GMagician. My M43 is above, but the problem is the same.
Then I downloaded Marlin 1.1.9 again and make only the minimum changes in Pins and activated dual-Z-Axis.
Same problem.
I changed the microswitches so i must not invert them: same problem.

If I manually trigger Z-Endstop or Z2-endstop M119 give's correct results.

So I don't know what I can do.

MarphXL, 2018-10-11:
I forget: With the new microswitches I make new cables and I make them VERY short. And as I wrote manually trigger-test with M119 are all correct.

GMagician, 2018-10-11:
Don't know if already asked, but are you using bugfix-1.1.x? maybe is better to start with it and not with stable 1.1.9 (bugs have been solved meantime)

MarphXL, 2018-10-11:
The first try was the bugfix 1.1.9, then stable 1.1.9. 5 days ago thinkihead said I should try bugfix 2.0 and I tested it with bugfix 2.0. Same problem. So what could it be?
the error must be somewhere in my hardware. I can google, and/but nobody else has that mistake.
Is there a possibility to debug that dual-z-homing or could I insert some code?

MarphXL, 2018-10-11:
Today I tested twice other versions of the tmc2208stepper-Lib (0.21, 0.25). but nothing changes

MarphXL, 2018-10-13:
Everything goes OK with Marlin 1.1.7. But 1.1.9 has a bug.

MarphXL, 2018-10-13:
I Try it with 1.1.9 with **absolutely** same configuration.h, configuration_adv and pins_RAMPS.h
I only must change 
```
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
#define Z2_DRIVER_TYPE TMC2208
#define E0_DRIVER_TYPE TMC2208
```

Result: with bf1.1.9 it doesnt work, but with 1.1.7 everything is fine and perfect

MarphXL, 2018-10-13:
Tried with 1.1.8 and this is also OK. 
So only Marlin 1.1.9 and Marlin bugfix 1.1.9 has the problem with DUAL-ENDSTOPS.

GMagician, 2018-10-13:
may you try to change Z2 input to `_ZMAX_` instead of `_YMAX_` ?

GMagician, 2018-10-13:
> Did you read that the problem doesn't exist at 1.1.7 and 1.1.8

Yes I read it, in version 1.1.9/2.0.x have been changed a lot of things about endstops logic, but as I told you my printer works correctly then it's not a so visible bug, don't you think so?

GMagician, 2018-10-13:
Try to enable `PINS_DEBUGGING` and post the serial log

MarphXL, 2018-10-15:
Hi GMagician, look above the 4th post.

GMagician, 2018-10-15:
I don't meant to see M43 echo, but echo while homing

thinkyhead, 2018-10-25:
I've been unable to determine what causes this issue so far. Since endstops underwent an overhaul for 1.1.9, maybe @ejtagle can provide more insight.

ApplePaeppel, 2018-10-29:
I have the same problem. 
Confirmation: works in 1.1.8 but not in 1.1.9

ejtagle, 2018-10-29:
set

```#define ENDSTOP_NOISE_THRESHOLD 2```

And try again...

MarphXL, 2018-10-30:
I didn't see any difference in the loggings, but I see that the 1st try goes good and leveled correctly, but the 2nd try didn't hit the right endstop at any time

MarphXL, 2018-10-30:
> > Did you read that the problem doesn't exist at 1.1.7 and 1.1.8
> 
> Yes I read it, in version 1.1.9/2.0.x have been changed a lot of things about endstops logic, but as I told you my printer works correctly then it's not a so visible bug, don't you think so?

hmm. Did you try to "dis"-level the axis, so you can test the two situations:
1. right side hit the endstop first
2. left side hit the endstop first

If this work for your printer with actual 1.1.9bugfix then I give you a beer.

GMagician, 2018-10-30:
@MarphXL 

> If this work for your printer with actual 1.1.9bugfix then I give you a beer.

is this valid for version 2.0? 'cause on my printer it works perfectly ;-)

Please enable `PINS_DEBUGGING`, then `M43 E1` and then `G28` and finally post output. This new output should report also endstop detectged changes

MarphXL, 2018-10-31:
i only want to print with dual-endstops and tmc2208. if this works on marlin 2.0 i will send you the beer 😀
i tried in the past with 2.0 but it did not work.
I will try with m43 e1 today and post results

MarphXL, 2018-10-31:
did you have tmc2208? perhaps that is the reason?

GMagician, 2018-10-31:
No I don't have tmc2208 but endstops logic is not stepstick dependent (only sensorless home may change behaviours)

MarphXL, 2018-10-31:
Here are the results with M43 E1 and please recognize that the printer doesn't hit the right endstop in the first try (+++ hit left endstop first):

+++ Hit left side first +++++++++++
```
Send: G91
 G28 Z0
 G90
[...]
Recv: >>> G28
Recv: Machine Type: Cartesian
Recv: Probe: NONE
Recv:   current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: >>> homeaxis(Z)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -600.00, [4.00])
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, 2.00, [4.00])
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   Z_MIN:1
Recv: 
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, -4.00, 1.00)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
[...]
Recv:   Z_MIN:0
Recv: 
Recv: <<< do_homing_move(Z)
Recv: >>> set_axis_is_at_home(Z)
Recv: For Z axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 400.00
Recv: > home_offset[Z] = 0.00
Recv:   current_position=(0.00, 0.00, 0.00) :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Z)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 0.00) : clean_up_after_endstop_or_probe_move
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: <<< G28
[...]
Recv:   Z2_MIN:0
Recv: 
Recv:   Z2_MIN:1
Recv: 
[...]
Recv:   Z2_MIN:0
Recv: 
Recv:   Z2_MIN:1
Recv: 

```

+++ Hit right side first +++++++++++
```
Send: G91
 G28 Z0
 G90
[...]
Recv: >>> G28
Recv: Machine Type: Cartesian
Recv: Probe: NONE
Recv:   current_position=(0.00, 0.00, 30.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: >>> homeaxis(Z)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -600.00, [4.00])
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
[...]
Recv:   Z2_MIN:0
Recv: 
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, 2.00, [4.00])
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   Z_MIN:0
Recv: 
Recv:   Z_MIN:1
Recv: 
Recv:   Z2_MIN:1
Recv: 
[...]
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, -4.00, 1.00)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   Z2_MIN:0
Recv: 
[...]
Recv:   Z_MIN:0
Recv: 
Recv: <<< do_homing_move(Z)
Recv: >>> set_axis_is_at_home(Z)
Recv: For Z axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 400.00
Recv: > home_offset[Z] = 0.00
Recv:   current_position=(0.00, 0.00, 0.00) :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Z)
Recv:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 0.00) : clean_up_after_endstop_or_probe_move
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: <<< G28
```
Here is a compare of the relevant passage:
[image]

MarphXL, 2018-10-31:
btw: I cannot compile actual bugfix 2.0 : Error "filenames are too long" but I cant change them.

GMagician, 2018-10-31:
Right case is ok, on left It seems endstop trigger comes at homing end....No idea!

thinkyhead, 2018-11-01:
> I cannot compile actual bugfix 2.0

To compile 2.0.x on Windows will require Arduino 1.9 beta or PlatformIO.

GMagician, 2018-11-01:
I think that right is what should be expected, while left... there are missing Z2 triggers that appears just too late... can't explain why?

GMagician, 2018-11-01:
ops wait a moment...first bump is correct and second is not or both are not good? I see here that after first bump it raise but endstop Z2 is not released...

MarphXL, 2018-11-01:
left side both are wrong; right side both are good

MarphXL, 2018-11-01:
I want to know what is the difference between your printer and mine?
Did you have all Endstops at interrupt-pins?

GMagician, 2018-11-01:
Well this is my printer:
Marlin 2.0 + mega2560 + ramps 1.4 + drv8825
Interrupts enabled and Z2 uses `_YMAX_`

thinkyhead, 2018-11-30:
Let's try using `M43 E1` without doing any homing.

- Raise the Z axis so no endstops are triggered.
- Send `M120` to turn on all the endstops.
- Send `M43 E1` to start monitoring endstops.
- Press and release the left (Z) endstop.
- Press and release the right (Z2) endstop.
- Press and hold the left (Z) endstop, then press and release the (Z2) endstop.
- Press and hold the right (Z2) endstop, then press and release the (Z) endstop.

Do you see anything unusual going on in this scenario?

joergles, 2018-12-08:
Did you try switching the endstop connectors of the two Z-endstops on the board ? It worked for me with the endstop for Z stepper on E1 using ZMIN and the stepper on Z using ZMAX (the additional one). Whyever ... I use Marlin 1.1.9

MarphXL, 2018-12-15:
[image]

@thinkyhead 
I tested it, but everything looks quite good

@joergles 
thanks for the tipp, but I tried this before with zmax and with ymax and I tried another board, but with no effects

MarphXL, 2018-12-15:
ah my mistake. compilation is ok

boelle, 2019-02-21:
@MarphXL 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

MarphXL, 2019-02-21:
it is a bug in the handling of interrupts. Not every interrupt-pin works and in source code you coulg chamge the endstop-passage

boelle, 2019-02-21:
and you are using bugfix 2.0?

MarphXL, 2019-02-21:
of course two printers with 2.0. and ¹.1.9 on another printer.

boelle, 2019-02-21:
funny, i also use dual z drivers and dual z endstop, no problems

how old is the 2.0 copy you have?

MarphXL, 2019-02-22:
funny? it is frustrating. i guess that you did not use tmc2208 with serial connection? 
the rx for every tmc2208 needs an interrupt.
dual z-endstop interrupts and the tmc-interrupts... this combination is buggy

boelle, 2019-02-22:
how old is the 2.0 copy you have?

MarphXL, 2019-02-22:
5 weeks. I installed every month a new bugfix since September 2018

GMagician, 2019-02-22:
I don't think it's a buggy combination (even if it's possible) but a congestion combination. You are using a 2560 board that has to handle a lot of things and maybe your real issue is that board has not enough computation power to handle all of these things.

boelle, 2019-02-22:
5 weeks is also very old... 2.0 is updated every 2nd day

and 2560 is a very SLOW board, try re-arm

[10 more replies]

closed 2019-07-11 by shitcreek