MarlinFirmware/Marlin #12025: mks genL not recogized

baharmark, 2018-10-07:
when pressing verify I get error unknown mother board value set configuration.h.
this is the #define BOARD_MKS_GEN_L         53    // MKS GEN L

thinkyhead, 2018-10-07:
Make sure your `Configuration.h` says:
```cpp
#define MOTHERBOARD BOARD_MKS_GEN_L
```
…and not something odd like…
```cpp
#define BOARD_MKS_GEN_L
```

baharmark, 2018-10-07:
ok it just has board because that was in board.h

closed 2018-10-07 by thinkyhead

baharmark, 2018-10-07:
and did not change.

thinkyhead, 2018-10-07:
Please ZIP and attach your `Configuration.h` and `Configuration_adv.h` files to your next post.

baharmark, 2018-10-07:
I made no changes to adv
[Documents.zip](https://github.com/MarlinFirmware/Marlin/files/2453574/Documents.zip)

thinkyhead, 2018-10-07:
Um… Please try again. Your `Configuration.h` is full of garbage.
[image]

baharmark, 2018-10-07:
hope this better
[Documents.zip](https://github.com/MarlinFirmware/Marlin/files/2453580/Documents.zip)

thinkyhead, 2018-10-07:
As already pointed out, your `MOTHERBOARD` definition is wrong. It should be:
```cpp
#define MOTHERBOARD BOARD_MKS_GEN_L
```
…not…
```cpp
#define MOTHERBOARD BOARD_MKS_GEN_L         53    // MKS GEN L
```

baharmark, 2018-10-07:
thank you that did it.