MarlinFirmware/Marlin #12030: Z drops down into part while printing

serhattural, 2018-10-07:
I checked gcode. There is no line about lower Z. But after some layers Z goes down. I added a gcode. 
[ZDownBug.zip](https://github.com/MarlinFirmware/Marlin/files/2453982/ZDownBug.zip)

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2018-10-07:
Try the new branch update https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x || https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x || https://github.com/MarlinFirmware/Marlin/tree/1.1.x

boelle, 2019-02-21:
@serhattural problem solved in bugfix 2.0?

closed 2019-05-16 by thinkyhead