MarlinFirmware/Marlin #12037: [Bugfix-2.0.x] EEPROM CRC mismatch after 2 printer reset

F0x06, 2018-10-08:
### Description

When i reset my printer 2 times i got `Error:EEPROM CRC mismatch - (stored) 32608 != 3663 (calculated)!`

### Steps to Reproduce

1. Restore fail-safe
2. Configured my printer (Delta calibration etc...)
3. Save settings (M500)
4. Power-off the printer
5. Power-it back on
6. Connect to serial (When connects it restarts the printer, this is the second reset)
Then i got 

PS: If i don't use serial and just power of the printer 2 times i get the same result

```
Connecting...
start
Printer is now online.
echo:Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: @f0x06
echo:Compiled: Oct  8 2018
echo: Free Memory: 3028  PlannerBufferBytes: 1472
Error:EEPROM CRC mismatch - (stored) 364 != 28739 (calculated)!
[ERROR] Error:EEPROM CRC mismatch - (stored) 364 != 28739 (calculated)!
```

**Expected behavior:** Keep my EEPROM settings

**Actual behavior:** EEPROM CRC mismatch and can't recover my settings

#### Additional Information

Version: d5017e2788f042d5b149a1e16dcbdc672d092015
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2457229/Configs.zip)

mrambossek, 2018-10-08:
This sounds like the problem I had, see https://github.com/MarlinFirmware/Marlin/issues/12015#issuecomment-427556710

F0x06, 2018-10-08:
@gmrambossek Thanks for the ref ! :)

thinkyhead, 2018-10-08:
Uncomment this option in `configuration_store.cpp` and the read/write functions will give us a better indication of which fields in the EEPROM read/write are misbehaving.
```cpp
// Check the integrity of data offsets.
// Can be disabled for production build.
//#define DEBUG_EEPROM_READWRITE
```

F0x06, 2018-10-08:
@thinkyhead I'll try this when I can, maybe tomorrow 👍

closed 2018-10-23 by F0x06