MarlinFirmware/Marlin #12045: G29 "E" option not available when tilting mesh

bluehallu, 2018-10-09:
My Bltouch accuracy is heavily compromised if it's not stowed after each probe:

`>>> m48 p50 V2 E
Mean: 0.036558 Min: -0.004 Max: 0.058 Range: 0.063`

`>>> m48 p50 V2
Mean: -0.049575 Min: -0.259 Max: 0.034 Range: 0.294`

However, E parameter is not available for "G29 J". This means even though accuracy is good for repeatability test and automatic mesh creation "G29 P1", accuracy is crap when I actually level before a print through my startup g-code.

gloomyandy, 2018-10-09:
If you have a good mesh stored then you really shouldn't need to use G29 J before each print, just load the mesh and use it. But yes it would make sense to be able to use the E option for all multiple probe operations. 

Oh and when reporting things it really helps if you post your configuration. There are several different types of bed levelling available in Marlin that have different G29 options (but which include P1, J and E options) I'm guessing that you are using UBL?

bluehallu, 2018-10-09:
@gloomyandy I've updated to mention UBL, yes. I think the idea of tilting the mesh before print (as recommended on the UBL guide http://marlinfw.org/docs/features/unified_bed_leveling.html) is that it will correct for beds that are hard to keep level while still reusing the mesh to compensate for small defects in the bed surface.

boelle, 2019-03-13:
@bluehallu is this still an issue with latest bugfix 2.0?

closed 2019-05-16 by thinkyhead