MarlinFirmware/Marlin #12046: [FR] Configurable power-loss-recovery

timoxd7, 2018-10-09:
### Description

Ability to configure the power-loss-recovery. At the moment, it only allows to save the state at every layer start. It would be great to change it to 'on every layer start' and additionally 'after a set traveled distance' or 'after a given amount of commands' and the ability to specify this in Configuration_adv.h. A configurable filename could also be helpful.

Very large prints with low layer count but high layer print time would benefit from such a feature. (i dont' know how it is saved on SD, but i think it is just a single variable which has to be changed in the File on SD while saving at printtime).

thinkyhead, 2018-10-09:
I’m in the process of overhauling PLR to reduce SRAM usage, to be able to enable / disable it, and to make it more configurable. As a new and semi-experimental feature, some of the options you mention are already in the current version as “hidden” options in the code. They will be promoted to the Configurations as the feature is further developed.

closed 2018-11-05 by thinkyhead

thinkyhead, 2018-11-16:
See #12440 for an updated and optimized version of Power-Loss Recovery. Needs testing. Enable the `DEBUG_POWER_LOSS_RECOVERY` option in `power_loss_recovery.h` to get extra output to the serial console when recovering. (You should be able to connect a host when the "Resume"/"Cancel" menu appears and it will go back to recovery mode when it reboots.)