MarlinFirmware/Marlin #12047: Implement "standard" G0 command

comps, 2018-10-09:
(This is a spiritual successor to #256, which I cannot reopen.)

Please implement a more "standard" (LinuxCNC / grbl / etc.) version of the G0 command, which is/was intended for a ["rapid positioning" move](https://en.wikipedia.org/wiki/G-code#List_of_G-codes_commonly_found_on_FANUC_and_similarly_designed_controls_for_milling_and_turning), meaning "as fast as the machine allows". In Marlin world, this would presumably be at `DEFAULT_MAX_FEEDRATE`, ignoring any previous or in-line `F` commands.

These "rapid" moves are not intended for the actual cutting/printing operations, but rather for re-positioning of the spindle/printhead, such as X/Y travel moves or "Z-hop" (or spindle retract).

Assuming some slicers use `G0` with a feedrate specified where they should really use `G1`, this feature should probably be optional, in the spirit of #12013.

thinkyhead, 2018-10-09:
Only SCARA machines can safely implement a G0 fast move differently from G1. We are already discussing in another thread whether to override the F feedrate, and generally the consensus is that it’s not a great idea in Marlin. But of course if someone wants this for their CNC system we can make it an option for those users. But it won’t be the standard thing for 3D printing, and we won’t ask slicers to change their assumptions for 3D printing. They should continue to use G0 Fnnn (if they already do) for non-printing moves.

comps, 2018-10-10:
Please note that it doesn't necessarily **have to** move differently than G1, for many CNC (and 3D printing) setups it's really just about ignoring the specified feedrate and interpolating like G1. Some more advanced CAMs allow you to choose different (spindle) retract modes based on how G0 behaves on that particular machine (if it does straight interpolated movement, you can get away with much less retract).

I made this issue for compatibility reasons with Fusion360 and other CNC routing CAMs, not really due to ideological or standardization reasons.

Thanks!

TheSFReader, 2018-10-10:
thinkyhead, two questions :
- I see in some cases, there is a planner.synchronize at the end of the G0-G1 move. would that help mitigate resonances if one were to use a "fast" G0 ? likewise, would S-Curve acceleration help there ?
- if one were to implement this suggestion via updating the feedrate in the G0_G1 before prepare_move_to_destination and restoring it after the call, do you think there could be cases where the "G0 feedrate" could have effect outside of this ? (Emergency parser or other preemptive cases ?)

closed 2018-10-26 by thinkyhead