MarlinFirmware/Marlin #12048: [Potential BUG] Error:EEPROM datasize error

boelle, 2018-10-09:
decided it was time to update to latest bugfix 2 and i used winmerge to compare the config files so i could port over my old settings

for some reason i cant get it to same to eeprom i get: 

**Error:EEPROM datasize error**

config files attached

using ramps board, dual z endstops and dual z drivers. 

should i clear the eeprom?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2460644/Configuration.zip)

boelle, 2018-10-09:
i tried the clear eeprom sketch found within arduino IDE, still the same

right after connect i get:

Recv: echo:EEPROM version mismatch (EEPROM= Marlin=V60)
Recv: ok
Send: M501
Recv: echo:EEPROM version mismatch (EEPROM= Marlin=V60)
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]

so it has to do with version i guess

boelle, 2018-10-09:
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (629 bytes; crc 63216)
Recv: Error:EEPROM datasize error.
Recv: ok
[...]

boelle, 2018-10-09:
one more odd thing i just noted

i can move Z down, but not up

ie i can home and since i have max z endstops it moves up like it should and does the retouch like with x and y axis

and then i can issue G0 Z1 and it goes down

but if i then issue G0 Z2 it does not move up again

boelle, 2018-10-09:
and when i compile i sometimes get this, and then i just compile again

sketch\src\module\temperature.cpp: In function 'print_heaterstates':

sketch\src\module\temperature.cpp:2402:3: internal compiler error: Segmentation fault

   }

   ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\boher\Desktop\Firmware\Arduino IDE 1.9.0\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/boher/desktop/firmware/arduino ide 1.9.0/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

boelle, 2018-10-09:
also the extruder is kind of acting up

will try and load an version from 29/7 with same config

Ludy87, 2018-10-09:
Try this https://github.com/MarlinFirmware/Marlin/issues/12037#issuecomment-42796525 and write your result

boelle, 2018-10-09:
will do later

but now trying the version i got 29/7 and at least i can now move z both up and down without any issue

dinner soon and will try #12037 with the version from today once i get back

thinkyhead, 2018-10-09:
To get more information about where the EEPROM error lies you can enable the `DEBUG_EEPROM_READWRITE` option in `configuration_store.cpp` and re-flash. On `M501` and `M500` this will print a message to the console indicating which field in the EEPROM is the wrong size.

boelle, 2018-10-09:
will do so in a an hour or so.... just switched to direct drive and dialing that in before i mess more with firmware

but i can say as much the same config with the version from 29/7 works

would it be wise to upload both new and old config files? or does it not matter that much?

thinkyhead, 2018-10-09:
Regardless of configuration there should be no EEPROM errors.

boelle, 2018-10-09:
oki, i just thought that it could be nice to compare configs, but then again i have used winmerge to compare the 2 config files so they should be the same

not sure if i get arround to flash the firmware today, its 1am and even the best need some sleep :-D

but it will be the first thing i do in the morning

thinkyhead, 2018-10-10:
Change `EEPROM_OFFSET` to 100 and try again. This might fix the issue.

boelle, 2018-10-10:
also in configuration_store.cpp?

thinkyhead, 2018-10-10:
Yes. That's where `EEPROM_OFFSET` lives.

Already patched in `bugfix-2.0.x`.

boelle, 2018-10-10:
oki.. but will try it and confirm first.

boelle, 2018-10-10:
still getting this

Send: M504
Recv: echo:EEPROM version mismatch (EEPROM=ERR Marlin=V60)
Recv: ok
Send: M501
Recv: echo:EEPROM version mismatch (EEPROM=ERR Marlin=V60)
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]

rmoravcik, 2018-10-10:
I just tested https://github.com/MarlinFirmware/Marlin/commit/b2c1cd7eda5aaec3ec9ef5322e9502025ce2a2f2 and it's working fine. The version from 7th of october wasn't working for me too

boelle, 2018-10-10:
well i will give the lastest bugfix 2 another go tomorrow

iamaldovanucci, 2018-10-10:
Having the same issue with the MKS SBASE. Seems to be related to the display and or SD Support. All I have to test with is a Full Graphic Display but if I disable the display and SD card EEPROM works fine. This is my first bug report so I'm not entirely sure what else to share. Just did a fresh build of b2c1cd7 in VS Code with platform.io to make sure I didn't goof something up.

boelle, 2018-10-10:
@iamaldovanucci 

you can be right, see the error i got: Recv: Error:Field lcd_preheat_hotend_temp mismatch.

and in the first post you can see my config, and i too use the full graphic display

tomorrow when i have time i will try and disable the display and see how that goes, but it would then mean i cant use the sd card and i need that

one leasson learned is to have a copy of what used to work :-D

JohnnyTheOne, 2018-10-11:
I have the same problem

18:52:32.256 : N16 M502*19
18:52:32.256 : echo:Hardcoded Default Settings Loaded
18:52:38.818 : N17 M501*17
18:52:38.818 : Error:EEPROM datasize error.
18:52:38.818 : echo:Index: 537 Size: 569
18:52:38.834 : echo:Hardcoded Default Settings Loaded
18:52:44.850 : N18 M504*27
18:52:44.850 : Error:EEPROM datasize error.
18:52:44.850 : echo:Index: 537 Size: 569

JohnnyTheOne, 2018-10-14:
I have enabled #define NO_VOLUMETRICS and the problem is gone

boelle, 2018-10-14:
oki, will try this

i totally forgot it as i'm still trying to dial in my printer, but think my part cooling fan duct does not provide enough cooling

will start by downloading a fresh copy and then port over my old settings

boelle, 2018-10-14:
right out the box i get these compile errors, but i guess those are related to the compiler and not my config. is remember correct @thinkyhead once said that comments after a pin number needs to be removed. will try removing the comment on this one

   #define CONTROLLER_FAN_PIN 4        // Set a custom pin for the controller fan

In file included from c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\hal.h:38:0,

                 from sketch\src\inc/MarlinConfig.h:30,

                 from sketch\src\Marlin.h:25,

                 from sketch\src\Marlin.cpp:31:

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\configuration_adv.h:201:39: error: pasting "/* Set a custom pin for the controller fan*/" and "_DDR" does not give a valid preprocessing token

   #define CONTROLLER_FAN_PIN 4        // Set a custom pin for the controller fan

                                       ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\fastio_AVR.h:84:42: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                          ^

sketch\src\Marlin.cpp:847:5: note: in expansion of macro 'SET_OUTPUT'

     SET_OUTPUT(CONTROLLER_FAN_PIN);

     ^

sketch\src\Marlin.cpp:847:16: note: in expansion of macro 'CONTROLLER_FAN_PIN'

     SET_OUTPUT(CONTROLLER_FAN_PIN);

                ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\configuration_adv.h:201:39: error: pasting "/* Set a custom pin for the controller fan*/" and "_PIN" does not give a valid preprocessing token

   #define CONTROLLER_FAN_PIN 4        // Set a custom pin for the controller fan

                                       ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\fastio_AVR.h:84:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                                             ^

sketch\src\Marlin.cpp:847:5: note: in expansion of macro 'SET_OUTPUT'

     SET_OUTPUT(CONTROLLER_FAN_PIN);

     ^

sketch\src\Marlin.cpp:847:16: note: in expansion of macro 'CONTROLLER_FAN_PIN'

     SET_OUTPUT(CONTROLLER_FAN_PIN);

                ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

boelle, 2018-10-14:
```

In file included from c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\hal.h:38:0,

                 from c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\inc\marlinconfig.h:30,

                 from sketch\src\module\stepper.h:46,

                 from sketch\src\module\stepper.cpp:80:

sketch\src\module\stepper.cpp: In static member function 'static void Stepper::set_directions()':

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\configuration.h:913:22: error: 'flse' was not declared in this scope

 #define INVERT_Y_DIR flse

                      ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\fastio_AVR.h:69:7: note: in definition of macro '_WRITE_NC'

   if (V) SBI(DIO ## IO ## _WPORT, DIO ## IO ## _PIN); \

       ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\fastio_AVR.h:91:31: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V)           _WRITE(IO,V)

                               ^

sketch\src\module\stepper_indirection.h:149:30: note: in expansion of macro 'WRITE'

   #define Y_DIR_WRITE(STATE) WRITE(Y_DIR_PIN,STATE)

                              ^

sketch\src\module\stepper.cpp:280:28: note: in expansion of macro 'Y_DIR_WRITE'

   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v)

                            ^

sketch\src\module\stepper.cpp:323:7: note: in expansion of macro 'Y_APPLY_DIR'

       A##_APPLY_DIR(INVERT_## A##_DIR, false); \

       ^

sketch\src\module\stepper.cpp:323:21: note: in expansion of macro 'INVERT_Y_DIR'

       A##_APPLY_DIR(INVERT_## A##_DIR, false); \

                     ^

sketch\src\module\stepper.cpp:335:5: note: in expansion of macro 'SET_STEP_DIR'

     SET_STEP_DIR(Y); // B

     ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\configuration.h:913:22: error: 'flse' was not declared in this scope

 #define INVERT_Y_DIR flse

                      ^

c:\users\boher\appdata\local\temp\arduino_build_134145\sketch\src\hal\hal_avr\fastio_AVR.h:75:7: note: in definition of macro '_WRITE_C'

   if (V) port_bits = ~port_bits;                            /* For setting bits, invert the mask */ \

[... 112 lines cut ...]
   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v)

                            ^

sketch\src\module\stepper.cpp:327:7: note: in expansion of macro 'Y_APPLY_DIR'

       A##_APPLY_DIR(!INVERT_## A##_DIR, false); \

       ^

sketch\src\module\stepper.cpp:327:22: note: in expansion of macro 'INVERT_Y_DIR'

       A##_APPLY_DIR(!INVERT_## A##_DIR, false); \

                      ^

sketch\src\module\stepper.cpp:335:5: note: in expansion of macro 'SET_STEP_DIR'

     SET_STEP_DIR(Y); // B

     ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

boelle, 2018-10-14:
well it compiles

test print is done in 1½ hours and i can upload after that, hunting down a cooling issue

boelle, 2018-10-14:
my config just in case it will be needed later on, i have the pins file for ramps included as i have added a dummy stepper and switched arround the thermistor inputs, for some reason they where not the same anymore and i dont want to touch the hardware if i can avoid it

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2476474/Config.zip)

boelle, 2018-10-14:
after flash and first boot:

```
Recv: echo:EEPROM version mismatch (EEPROM=V55 Marlin=V61)
Recv: ok
Send: M501
Recv: echo:EEPROM version mismatch (EEPROM=V55 Marlin=V61)
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
```

boelle, 2018-10-14:
```
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (629 bytes; crc 23185)
Recv: Error:EEPROM datasize error.
Recv: ok
[...]

```

will now try what @JohnnyTheOne suggests

Phr3d13, 2018-10-14:
I thought the option "Initialize EEPROM" in the LCD menus was for instances such as this?

boelle, 2018-10-14:
going back to my copy from 29-7-2018

of course at first boot it complains about eeprom version, but it also shows that it writes eeprom version to the eeprom, but no other errors and it only happens once

```
Send: M504
Recv: echo:EEPROM version mismatch (EEPROM=V61 Marlin=V55)
Recv: ok
Send: M501
Recv: echo:EEPROM version mismatch (EEPROM=V61 Marlin=V55)
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (679 bytes; crc 40706)
Recv: ok
[...]
```

gloomyandy, 2018-10-14:
Did you try enabling DEBUG_EEPROM_READWRITE? It might help to identify what is causing the problem.

boelle, 2018-10-14:
upps,,,, nope... will do in the morning

boelle, 2018-10-14:
just before i go to bed, here it is

```
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: Error:Field lcd_preheat_hotend_temp mismatch.
Recv: ok
[...]

```

boelle, 2018-10-19:
will try it later today

its mominlaw birthday so will be out most of the day

boelle, 2018-10-19:
so... downloaded a fresh copy from today and in the process of porting over config

the new section  * Stepper Drivers makes me a bit in doubt if i have done it right 

//#define X_DRIVER_TYPE  DRV8825
//#define Y_DRIVER_TYPE  DRV8825
//#define Z_DRIVER_TYPE  DRV8825
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE DRV8825
//#define Z3_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE DRV8825
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

i'm mostly worrid since i have dual z drivers

my config is for 2 extruders but one of them is just "virtual" so i can use that one to display chamber temp on the display, i know there is a chamber temp sensor pin but that one only displays temp on serial data and not in display. it makes it more easy in octoprint too the way i do it

boelle, 2018-10-19:
of course when compiling ther is still an error due to the compiler does like comments

so in configuration_adv

#define CONTROLLER_FAN_PIN 4        // Set a custom pin for the controller fan

becomes

#define CONTROLLER_FAN_PIN 4

boelle, 2018-10-19:
still getting errors but not the same as before

```
Recv: Error:EEPROM datasize error.
Recv: echo:Index: 625 Size: 637
Recv: ok
Send: M501
Recv: Error:EEPROM datasize error.
Recv: echo:Index: 625 Size: 637
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (625 bytes; crc 28132)
Recv: Error:EEPROM datasize error.
Recv: ok
[...]

```

new zip file attached with configuration.h, configuration_adv.h, configuration_store.cpp and pins_RAMPS.h
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2496680/configuration.zip)

dahuafschmied, 2018-10-19:
At your first posts you had a difference of 4 bytes (1x float) that would match to my bug. But now you have a 12 byte difference (3x float). It seams you have a different bug than mine.

boelle, 2018-10-19:
yeah, i guess i'm lucky that i have an old copy that works so i can print while we figure this one out

but at least i dont have the Error:Field lcd_preheat_hotend_temp mismatch more

boelle, 2018-10-19:
Same result when order is

M502
M500
M501

Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (625 bytes; crc 28132)
Recv: Error:EEPROM datasize error.
Recv: ok
[...]
Send: M501
Recv: Error:EEPROM datasize error.
Recv: echo:Index: 625 Size: 637
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]

boelle, 2018-10-19:
if i disable dual endstop

```
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (613 bytes; crc 31850)
Recv: Error:EEPROM datasize error.
Recv: ok
[...]
Send: M501
Recv: Error:EEPROM datasize error.
Recv: echo:Index: 613 Size: 625
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]

```

dahuafschmied, 2018-10-21:
just for test. try to disable AUTO_BED_LEVELING_BILINEAR. it is a default value that is before lcd preheat and has a default value with multi 3x floats. i whish you luck!

rmoravcik, 2018-10-21:
Hi @boelle, @thinkyhead, I think I found the problem.

If you have a look to the definition of z3_endstop_adj in SettingsDataStruct, z3_endstop_adj defined only if Z_TRIPLE_ENDSTOPS is enabled, but MarlinSettings::save() and MarlinSettings::_load() wanted to write/read dummy also if Z_TRIPLE_ENDSTOPS is disabled.

There is also an problem with definition of toolchange_settings in SettingsDataStruct.

boelle, 2018-10-21:
i can confirm its fixed

```
Recv: echo:EEPROM version mismatch (EEPROM=V55 Marlin=V62)
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored (633 bytes; crc 40581)
Recv: ok
[...]
```

boelle, 2018-10-22:
hmmm

for some reason i cant close my issues or create new ones, any idea why @thinkyhead ?

in any case i hit a new issue where print stops due to the z probe is triggered

since i use the pinda probe on my mendel90 with the mk42 bed there is no way arround this, it will remain triggered until about 1mm up (0,79+)

i have this in configuration_adv.h:

```
// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
```

sine the option is not enabled it should not have the endstops on when homing is done at start of print

a more interesting one:

```
  /**
   * This option allows you to abort SD printing when any endstop is triggered.
   * This feature must be enabled with "M540 S1" or from the LCD menu.
   * To have any effect, endstops must be enabled during SD printing.
   */
  //#define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED
```

so since this feature is not  enabled and the other is not i would expect to be clear

but my print is aborted due to a z endstop hit, since my Z endstops are top/max mounted and work fine i would rule them out, the only other endstop is the probe itself

my config is still the same as 3 days ago so they can be found by scrolling up
https://github.com/MarlinFirmware/Marlin/issues/12048#issuecomment-431428871

boelle, 2018-10-22:
hmmm... its not the same on every print so not sure what is going on

just started an 5 hour test print to dial in my retraction.... got  some of those very fine hair stringing

closed 2018-10-22 by boelle