MarlinFirmware/Marlin #12057: [2.0.x] Compiler warning, unused variable in configuration_store.cpp

marcio-ao, 2018-10-10:
Not a biggie, but I thought I would mention that I am getting the following compiler warning:

```
src/module/configuration_store.cpp: In static member function ‘static bool MarlinSettings::_load()’:
src/module/configuration_store.cpp:1359:31: warning: unused variable ‘tmc_stepper_current’ [-Wunused-variable]
         tmc_stepper_current_t tmc_stepper_current;
```

The undefined variable in question is actually [line 1353](https://github.com/MarlinFirmware/Marlin/blob/b2c1cd7eda5aaec3ec9ef5322e9502025ce2a2f2/Marlin/src/module/configuration_store.cpp#L1353) in standard marlin.

Don't really know how to fix this properly. Should the `_FIELD_TEST(tmc_stepper_current);` and `tmc_stepper_current_t tmc_stepper_current;` be brought inside the `#if HAS_TRINAMIC`?

What does _FIELD_TEST do? I see it a lot in "configuration_store.cpp".

thinkyhead, 2018-10-10:
`_FIELD_TEST` is enabled with `DEBUG_EEPROM_READWRITE` and it tests to make sure that the current piece of data in the EEPROM about to be read or written has the correct offset, according to the `SettingsData` structure.

thinkyhead, 2018-10-10:
Ok, patched the compile warning.

I've been following a guideline set early in Marlin development, which is to read and write all EEPROM data that could exist, even for features that are disabled (using placeholder data on write) so that features can be enabled or disabled without any change to the EEPROM data structure. This was also a practical need in the past because the EEPROM data wasn't validated before being loaded into the SRAM settings, so garbage could be read into settings. The only remedy then was  to call `settings.reset` if the data was bad on `settings.load`. Now with the added validation step the settings will only be altered if the EEPROM data is verified to be good, so we don't need to call `settings.reset` on error; existing modifications to settings in SRAM will remain as they are.

All that said, it's still not quite perfect. Changing the number of extruders and reflashing, for example, will always require the EEPROM to be reset. The EEPROM data size could be reduced quite a bit if we were willing to sacrifice the placeholders for disabled settings and apply triage to the settings we care about most. For example, requiring an EEPROM reset if users change their stepper drivers to TMC2130s, add/remove a filament sensor, etc. Keeping placeholders also makes the code a bit more complex and harder to maintain, and it's an open question whether it's all that important to reduce the need for EEPROM reset when flashing new configurations.

tcm0116, 2018-10-11:
>Keeping placeholders also makes the code a bit more complex and harder to maintain, and it's an open question whether it's all that important to reduce the need for EEPROM reset when flashing new configurations.

One option could be to use a bit field to indicate which elements are actually present in the EEPROM data. This could have the added benefit of allowing the EEPROM data to be migratable if the data structure changes. Of course, certain rules about how the data structure can be changed would have to be enforced for that to be effective. But, it would be nice to not have to re-calibrate and re-level when upgrading Marlin.

teemuatlut, 2018-10-11:
> it would be nice to not have to re-calibrate and re-level when upgrading Marlin

Bit of a chore to do but you can export the UBL commands that determine the point values as a gcode. You can then run it back to the machine after upgrading. I think `G29 S-1` is the command to report everything.
Not really a solution in this discussion but perhaps something that can help you.

closed 2018-10-15 by marcio-ao