MarlinFirmware/Marlin #12058: Home X and Y axis only after Z axis moved (not acting like Z home)

Yoonutz, 2018-10-10:
Please help me with thos problem. 
I have configured marlin with 4 TMC2130 drivers and everything works fine during movement of the axis, but when I want to home X or Y axis, first thing Z moves down (My bed is moving up or down ( upper position is home)) and it down not act like Z home perse, it stays down, and only after Z has done it's move, my X or Y axis is homing. I am using optical endstops.

What it might be the cause? I get no errors what so ever. Ramps board changed twice. Swapped all tmc2130 drivers. I only tried configuring it with X axis only, and nothing else installed on the board. Nothing did changed. My best guess it's that is something I changed in software. Trying to configure  Marlin 1.1.9 

Thank you,

ManuelMcLure, 2018-10-10:
This is probably due to `Z_CLEARANCE_BETWEEN_PROBES` - Marlin will by default move Z 5mm before homing X and Y to avoid any bed clips or such.

You can override this by explicitly setting `Z_HOMING_HEIGHT` to 0, but the question becomes why you want to disable this. It's a safety feature that should not affect the final homing results.

Yoonutz, 2018-10-11:
Well, I dont think they will solve my problem because, everytime I plug the printer, on first try it does this, then everyting is all fine if I home every axis by itself. It still keeps these weird movements when I press home XYZ button

closed 2018-10-17 by Yoonutz

thinkyhead, 2018-10-17:
Some beds fall down under gravity, so we added this option for those cases:
```cpp
//#define UNKNOWN_Z_NO_RAISE // Don't raise Z (lower the bed) if Z is "unknown." For beds that fall when Z is powered off.
```