MarlinFirmware/Marlin #12062: TMC2208 Noisy/grinding when connected to UART

MotoGP1199, 2018-10-11:
### Description

I recently changed only one of my stepper drivers on the X axis (TMC2208) from legacy mode over to run on UART to test it out.  When I did that the stepper now makes a loud grinding/rattling noise when doing fast movements only.  I have hybrid mode and stealth chop enabled.
[Marlin-bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2467330/Marlin-bugfix-1.1.x.zip)

I have a 2130 on my E0 with SPI and it works fine.

The TMC2208 (Watterott) is configures the same as this article. https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/

I have the UART pin from the 2208 hooked up to PIN 40(TX with 1k OHM resistor) and 63 (RX) on the MKS GEN L board.  I was running the current (for x axis) at both 800MA and 1000MA and it operates the same.

Marlin (the most current 1.1.x BUG FIX
Board. MKS GEN L V1.0

Configuration.h_____________________\

#define X_DRIVER_TYPE  TMC2208  
#define Y_DRIVER_TYPE  TMC2208_STANDALONE  
#define Z_DRIVER_TYPE  TMC2208_STANDALONE  
#define X2_DRIVER_TYPE A4988
#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130 
#define E1_DRIVER_TYPE A4988
#define E2_DRIVER_TYPE A4988
#define E3_DRIVER_TYPE A4988
#define E4_DRIVER_TYPE A4988
**

Configuration.h Advanced__________________

#if HAS_TRINAMIC

  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          1000  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

MotoGP1199, 2018-10-11:
Never mind I figured it out.  I was finally able to find some data on the steppers that I have.  I increased the MA to 1500 which got rid of the grinding.  Also Lowered the Hybrid threshold which helped make it run smoother.

closed 2018-10-11 by MotoGP1199

jimmyeao, 2019-01-04:
@MotoGP1199 How did you do this, I have the same issue is it a case of changing the 
#define X_CURRENT to 1100
What did you lower the hybrid threshold too?

Thanks