MarlinFirmware/Marlin #12064: 1.1.x Cannot send commands via USB when WIFI is attached to AUX

karabas, 2018-10-11:
### Description

Mega2560 /ramps 1.4
WIFI ESP8826 is attached to AUX as default serial and powered from 3.3 arduino pin.
WIFI works perfectly. 
I use Pronterface to send commands.

The problem:
When WIFI is powered ON and USB is connected to PC, I am able to connect Pronterface to printer via USB but NOT able to send any command via USB!  
So I forced to turn off WIFI module if  I want to communicate via USB.

gloomyandy, 2018-10-11:
The more information you can provide the more likely someone can help. So your config files, plus things like what display you have attached, any special pins you have in use, what version of Marlin you are using etc.

igorepi, 2018-10-12:
Both onboard Mega2560 USB-to-serial bridge and ESP8266 are connected to the same RXD/TXD UART Atmega pins so you can't use USB connection and ESP8266 simultaneously.

igorepi, 2018-10-17:
You don't have any choice - Marlin accepts g-code and sends responses through Atmega2560 UART0 only (when not printing from SD card). So you can connect only one device at a time - PC through USB (PC <-> USB connector <-> onboard Mega2560 USB-to-serial bridge <-> UART0), ESP8266 or MKS TFT32 (AUX1 connector <-> UART0).

On urelated note - ESP8266 is not 5V-tolerant and you have to use 5V<->3.3V level shifter to connect it to 5V-level Atmega 2560 RXD/TXD.

karabas, 2018-10-17:
> You don't have any choice - Marlin accepts g-code and sends responses through Atmega2560 UART0 only (when not printing from SD card). So you can connect only one device at a time - PC through USB [...]

I got  3.3v from Arduino for powering ESP8266. As for Rx Tx no shifters are really needed. Works perfectly.  

I jumped to Marlin 2 to use SERIAL_PORT_2 option.
Now both USB and serial 2 work with hosts

closed 2018-10-17 by karabas