MarlinFirmware/Marlin #12065: MKS GEN 1.4 + 5xTMC2208 Z2 no UART

andrebu89, 2018-10-11:
### Hello Everyone,

i popped in 5xTMC2208 prepared for UART. 
X,Y,Z1,E0 are working as expected.
R/Wr are fully functional.

Problem is on Z2. 
I can't figure out how to get RX working on Z2, because the only left RX/TX Pin Combinations not working.

Is someone here who can help me?

AnHardt, 2018-10-11:
5 x TMC + 1 for the host = 6 serials. On a chip with only 4 UARTS. Means 2 soft serials. That's (too) ambitiously!

teemuatlut, 2018-10-11:
Make sure the RX pins are interrupt capable.

teemuatlut, 2018-10-11:
I don't think there's inherently anything that would prohibit you from using all five. But it's not something I've tried myself.
You could try running two from HW Serial (I use serials 1 and 3) and three from SW UART. I know those two UARTS are accessible on a ramps board but I don't know about the MKS board.

andrebu89, 2018-10-11:
> I don't think there's inherently anything that would prohibit you from using all five. But it's not something I've tried myself.
> You could try running two from HW Serial (I use serials 1 and 3) and [...]

yeah that could be possible. THink i need to evaluate a little bit more into this. The MKS Should be same as RAMPS with a little Bunch of PINS changed. Will keep Updating this :)

boelle, 2019-02-21:
@andrebu89 problem solved?

andrebu89, 2019-02-21:
not really, but it works with 4x uart tmcs. printer is printing and im
happy :D

boelle, 2019-02-21:
if happy maybe click the green close bottom below

closed 2019-02-21 by andrebu89