MarlinFirmware/Marlin #12066: TEMP_STAT_LEDS & EVENT LED confusion

ruggb, 2018-10-11:
@thinkyhead This LED thing is confusing me. It appears as if the LED functions (TEMP & EVENT) are display dependent since STAT_LED_BLUE_PIN 35 is defined in RAMPS.h if a VIKI display is enabled, but that is not stated anywhere that I can see. Then RGB_LED_B_PIN 35 is defined if RGB_LED or RGBW_LED is enabled (not displays). That seems like it could present a conflict but I find no info on that. What did I miss?

So when I added my TEMP_STAT_LEDS, I also had to define the pins, but since I use a discount display there was no conflict and I defined them in-line in RAMPS.h using pin 5 & 6 as the LEDs are not part of my display. But there are no instructions RE that and it may become a problem for me in future builds.

Does this make sense and should more be done to relieve confusion and prevent future problems - or is it just my issue since, apparently, I am not using this as designed. If so what is the "proper" way for me to define these pins?

thinkyhead, 2018-10-26:
Defining the pins yourself is fine — and the general approach to any custom hardware. It just happens the VIKI display has its own.

boelle, 2019-02-21:
@ruggb problem solved?

ruggb, 2019-02-23:
no idea but I got it working. thks

closed 2019-02-23 by ruggb