MarlinFirmware/Marlin #12067: [question] Led commands

GMagician, 2018-10-11:
I'm looking to change some little things on `PRINTER_EVENT_LEDS` while tuning pid to restore previous led status.
I'm facing something I can't understand since I don't have any led strip and I really don't know how they work.

1) Method `leds.set_color` seems to allow more that one led strip type but when neopixel is installed and "sequence" is active then there is one situation where all other led strip are not updated
```cpp
      if (++nextLed >= pixels.numPixels()) nextLed = 0;
      return;
```

this "sequence" is only used when something is heating (bed or hotend) but why not to update all other strips when led is last?

2) When setting white to neopixel if calls `set_color` with one white constant that may not be real color value for neopixel (and a subsequent call is done with real color: `set_neopixel_color(pixels.Color(NEO_WHITE));` but again, previous call is already done with no "sequence" active...why not do something like this in `setcolor`? and remote neopixel call?
```cpp
  #if ENABLED(NEOPIXEL_LED)

    const uint32_t neocolor = !isSequence && !!incolor is white!! ? pixels.Color(NEO_WHITE) : pixels.Color(incol.r, incol.g, incol.b, incol.w);
    static uint16_t nextLed = 0;

    pixels.setBrightness(incol.i);
    if (!isSequence)
      set_neopixel_color(neocolor);
    else {
      pixels.setPixelColor(nextLed, neocolor);
      pixels.show();
      if (++nextLed >= pixels.numPixels()) nextLed = 0;
      return;
    }

  #endif
```

and again `NEOPIXEL_IS_RGBW` says that it has 4 color componets and `HAS_WHITE_LED` says there are 4 components but `#define HAS_WHITE_LED (ENABLED(RGBW_LED) || ENABLED(NEOPIXEL_LED))
` why not `#define` HAS_WHITE_LED (ENABLED(RGBW_LED) || ENABLED(NEOPIXEL_IS_RGBW)) with this there is no need for `NEO_WHITE` to be defined.

Thanks for your attention

thinkyhead, 2018-10-26:
The sequence option lights up one LED at a time on a strip, instead of all of them at once. Who has multiple types of LEDs? The number of LEDs on a strip is arbitrary, so setting the color of other LEDs based on the last one in the strip being set is equally arbitrary.

Code shouldn't use `ENABLED(NEOPIXEL_IS_RGBW)` — but just `NEOPIXEL_IS_RGBW`.

GMagician, 2018-10-26:
>  Who has multiple types of LEDs?

don't know but code allow such situation. If not allowed I'd expecting a `#If enabled(one) ... #else if enabled (two) .... #else if enabled (third)` and not a `#if enabled(one)...#endif ...#if enabled(two) ... #endif`

> Code shouldn't use ENABLED(NEOPIXEL_IS_RGBW) — but just NEOPIXEL_IS_RGBW.

ok but is there a reason with `HAS_WHITE_LED` doesn't include `NEOPIXEL_IS_RGBW`?

by the way I posted a PR with solution in point 2 and I'm waiting for someone with led strips to test it

thinkyhead, 2018-10-26:
> ok but is there a reason with `HAS_WHITE_LED` doesn't include `NEOPIXEL_IS_RGBW`?

It looks like `HAS_WHITE_LED` is just used to determine that 4 parameters are sent to the LED color functions. For RGBW they'll be 0,0,0,255 and for RGB they'll be 255,255,255,0. NEOPIXEL also gets an intensity value, and that has to follow after the W value.

Just the way it is. I don't know that it's worth an overhaul.

GMagician, 2018-10-26:
thanks for your attention.

closed 2018-10-26 by GMagician