MarlinFirmware/Marlin #12068: [2.0.x-bugfix] EEPROM datasize error

mgrl, 2018-10-11:
### Description

Found an EEPROM datasize error. I searched here a little and found I have to enable the debug-output to assist with futher information. This is what the monitor prints:

```
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Oct 11 2018
echo: Free Memory: 86504  PlannerBufferBytes: 1664
Error:Field parser_volumetric_enabled mismatch.
Error:EEPROM datasize error.
echo:Index: 633 Size: 665
echo:Hardcoded Default Settings Loaded
```

### Steps to Reproduce

1. Load a fresh marlin 2.0.x-bugfix copy (I loaded today evening)
2. configure marlin (I disabled volumetric by #define NO_VOLUMETRICS, because I want to use marlin for openpnp)
3. setup in platform-io to compile for arduino due with radds shield
4. upload and start the monitor
5. reset eeprom and store again doesn't help...

#### Additional Information

* [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2470240/Marlin.zip)

formadi, 2018-10-12:
I have a same problem.

Lyr3x, 2018-10-12:
Yep hppend to me since yesterday too

ikarisan, 2018-10-12:
Same here!

Resetted the EEPROM and also executed M502 and M500.
But after a reset, I am getting the same "EEPROM datasize" error.

ikarisan, 2018-10-12:
Connecting...
start
Printer is now online.
echo:Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: Shinji Ikari
echo:Compiled: Oct 12 2018
echo: Free Memory: 3352  PlannerBufferBytes: 1648
Error:Field parser_volumetric_enabled mismatch.
[ERROR] Error:Field parser_volumetric_enabled mismatch.

Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.

echo:Index: 525 Size: 557
echo:Hardcoded Default Settings Loaded

povlhp, 2018-10-13:
Compiled for Ender-3 with BLTouch. Error here as well.

viper93458, 2018-10-14:
Ouch, I am seeing this one too. Welcome to testing! :)

povlhp, 2018-10-14:
Fixed in 213e94b for me

facet02, 2018-10-14:
Got the same Error:EEPROM datasize error... on my MKS Gen 1.4

mgrl, 2018-10-14:
can confirm that #12086 fixes the issue reported initially here.
@facet02 load the current version and try again. if the error still occurs, it's not likely to be related and you might open another issue for that.

closed 2018-10-14 by mgrl