MarlinFirmware/Marlin #12070: Multiple probing and other options suggestion

sergiofagundes, 2018-10-11:
It would be great to have some options in LCD Menu to change at runtime instead of compiling and updating the board.

One example, I'd like to have for example the multiple probing selectable from a LCD Menu, I don't remember now but there're other options too that could be set up in lcd menu. Something like an Advanced Menu Options...

InsanityAutomation, 2018-10-11:
The down side is the memory it takes. Every adjustable variable takes memory. Static defines don't.  Plus the memory the menu item itself takes. 

Next is taking eeprom space into account. Every configurable value also needs to be stored to eeprom to persist on reboot.

For things that are seldom changed once flashed, it's just not worth the memory on tight avr boards. Now if someone wanted it enough to write a pr with a config option to enable extra config menus and determine approximate memory usage so users could decide if they wanted it, sure some 32bit users may want it. Even then the eeprom space may become a concern. Otherwise it's a very time consuming task with a very small base of users who would find it advantageous in my opinion.

thinkyhead, 2018-10-26:
These kinds of things are a pretty low priority right now, and yet another obscure configurable option to add to the configurations. But if someone sends us a PR we'd consider merging it.

sjasonsmith, 2020-11-10:
This has been idle over two years, and is short on specifics. While multiple probing is provided as an example, it is a very broad request to essentially make all settings accessible through the LCD. Given the plethora options available in Marlin, this is not practical, and should be considered for specific options if we have workflows which justify the effort needed.

closed 2020-11-10 by sjasonsmith