MarlinFirmware/Marlin #12074: [1.1.x] Ender 3 default homing speeds too violent for X, Y endstops?

swilkens, 2018-10-12:
@thisiskeithb - tagging since you contributed the original profile for this printer

### Description

Stock Marlin 1.1.9 with Ender 3 defaults results in fairly violent homing, causing the X and Y axis end switches to bottom out. See this video for the default behavior, notice the end stop switches from bottoming out completely:

Turn up the volume to hear the audible "clank" when it bottoms out:
https://www.youtube.com/watch?v=YJmIMP_TwwY

When reducing the homing speed, the homing is less violent and the end stop switches do not bottom out. In the video below I reduced the homing speed to 20*60 down from the original 50*60. I've also increased the divisor for the bump from 2 up to 10. See the result:

https://www.youtube.com/watch?v=UkRmVmKVAUU

You'll notice the end stop doesn't bottom out after the bump. See the image blow for a compare between the final homing positions:

[image]

### Proposed modification

Current relevant stock settings for homing speeds are as follows:

Configuration.h
`#define HOMING_FEEDRATE_XY (50*60)`

Configuration_adv.h
`#define HOMING_BUMP_DIVISOR { 2, 2, 4 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)`

I propose to change that to:

Configuration.h
`#define HOMING_FEEDRATE_XY (20*60)`

Configuration_adv.h
`#define HOMING_BUMP_DIVISOR { 5, 5, 4 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)`

thisiskeithb, 2018-10-13:
Not that I have much say, but I like this change :)

swilkens, 2018-10-14:
It is strange to me that this happens though, it seems almost as is the MCU isn't capable of responding quickly enough. Or perhaps the endstops just have very shitty noise, I'll have to hook up a scope to see.

For now though, do you see this behavior as well @thisiskeithb? Just so I don't make modifications for an issue that potentially only affects me.

thisiskeithb, 2018-10-14:
The homing sequence was violent on both of my stock Ender-3 printers (they were a few “versions” apart), but I’ve since swapped to MKS Gen-L boards and modified my Ender-3 printers beyond stock.

swilkens, 2018-10-15:
Alright, made the pull requests. Further testing showed that only adjusting `HOMING_FEEDRATE_XY` was enough to stop this from happening, `HOMING_BUMP_DIVISOR` doesn't have to be touched.

Lord-Quake, 2018-11-18:
I just assembled an Ender-3 Pro and "had" the same problem.
The crashes are so violent that the small end stops will eventually break and fail.
Changing to "HOMING_FEEDRATE_XY (20*60)" made homing safe for use.
I just checked the current version of 1.1.9 (18.11.2018).  It still has "(50*60)" in the config.
The Ender-3 will become a very popular printer, if not already, and many users will upgrade the Marlin firmware as I did.
**Under the circumstances I strongly recommend changing the the config file ASAP to avoid any grief to unsuspected users.**

swilkens, 2018-11-18:
The modification of he homing speed for this printer was applied to the bugfix-1.1.x and bugfix-2.0.x branches. The 1.1.9 version you downloaded was the release state which hasn't received updates since the day of its release. This is why you don't see the change.

If you want the bugfix-1.1.x branch, look here:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

Or bugfix-2.0.x:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

The only source we have for the creality factory default is their github, which also uses `50x60` as the default homing speed for their printers:

https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/Configuration.h#L407

So do we break factory defaults or do we prevent potential damage? I would be inclined to do the latter, but then we might ask some users if this is an issue on other creality printers as well before we do so. 

Perhaps they just used slightly other switches in some batches, who knows.

Lord-Quake, 2018-11-18:
I did not write my experience lightly. I've been using Marlin for several years.
Creality is not using 1.1.9 right? So IMO it would be pointless to talk about default settings.
The general user is not going to download future releases or builds.
They are going to do exactly what I did, go to the release page and download the official version.
I was lucky, my end stops still work. Others might not be or they will find the crashes are normal and might have to live with the consequences in the future.
So please for the sake of Ender 3 users, change the config ASAP.......

thinkyhead, 2018-11-19:
That change was already made, on October 16 82c5c85c0f

Lord-Quake, 2018-11-19:
@thinkyhead: I know but but it's not in in the current official 1.1.9. release. IMHO we need a hot-fix ASAP. That is what I'm trying to convey.

thinkyhead, 2018-11-19:
I see. Well 1.1.9 won't be getting any more changes or hotfixes or tweaks anymore, so it's going to just be stuck that way, sorry. Let's hope most users just download straight from the main (1.1.x branch) page instead of going to the releases page.

Lord-Quake, 2018-11-19:
OK. While I believe this is a very bad idea I can't blame myself for not trying....

And while I'm here.... I will nonetheless say thanks for the great work you and everyone else is doing on Marlin!

thinkyhead, 2018-11-19:
Thanks! We had to draw the line somewhere and start saying "no" at some point. You're not the first or the last to have a major pressing issue. I promise, if we get a plague of complaints from Ender 3 users I will be happy buy them all new endstops.

CTRLurself, 2019-02-20:
Think head; this issue still exists in an Ender 5 that I just put together yesterday. I haven't homed it again after it hit once until I can manually make this update to it.

fsqonbdk, 2019-05-17:
Running TMC2130s on an MKS Gen L on an Ender 3 and reducing HOMING_FEEDRATE to 20*60 triggers sensorless homing too early. You will need to increase M914, in my case I had to increase it to M914 X16 Y14.

tablatronix, 2019-11-07:
Is there also a way to soft start homing moves also? My bed starts up just as violently as it stops.