MarlinFirmware/Marlin #12075: [2.0.x-bugfix]  Power supply 2 error

ModMike, 2018-10-12:
### Description

Power turns off when power button is released. 

### Steps to Reproduce

Set:

`#define POWER_SUPPLY 2`

also tried:

`#define POWER_SUPPLY 1`

**Expected behaviour:**

Printer is off when plugged in.  Turns on when button is pushed.  Turns off when button is pushed again.

**Actual behaviour:** [What actually happens]

Printer is off.  Power turns on when button is pushed but immediately turns off when the button is released.

#### Additional Information

Creatbot DX using a Rumba controller derivative with momentary push button power switch wired to a relay.  Printer will shut off with M81.

Power supply is the ubiquitous Meanwell LRS-350-24

[image]
[image]
[image]

ModMike, 2018-10-12:
Could this have anything to do with SETUP_POWERHOLD?  If so where is it configured?

thinkyhead, 2018-10-13:
Marlin contributors are volunteers who work on the firmware in their spare time for fun. Most of us have busy lives and may be engaged with family, job, friends, or recreation on any given day. You should expect delayed responses from time to time of 72 hours or more. And, only contributors who actually have knowledge and experience with the issue feel qualified to answer. As for me, I’ve been quite busy following up on pull requests and simply haven’t looked at the general issue queue for the last few days.

thinkyhead, 2018-10-13:
Try `M42` and see if it gives the same behavior. Use `M43` to see if your power pin is shared with some other function.

ModMike, 2018-10-13:
@thinkyhead Please do not mistake the intent of my comment.  I was previously in a conversation in another thread concerning USB stick readers with some one from Lulzbot who had an opinion about working on non GPL compliant machines.

He ultimately was tincredibly gracious and helpful but I started to wonder if maybe I was breaking some unwritten rule, which is why I asked the question when I got no response.

I thought I had worded my question appropriately but apparently it came across as peevish.

I respect the work everyone does here and I contribute were my limited skills allow me to.  I also Patreon the project.  Admittedly it’s low but I will increase it once my business takes off.

I would like to sincerely apologize to you and the community if I came across as being owed a response.  That was NOT my intention.

ModMike, 2018-10-13:
@thinkyhead I have been working on this issue for close to a week and believe me when I say I tried everything, including disabling as many pins as possible in the pins-CREATBOT file I created to ensure that my M43 command would reveal if they were occupied elsewhere.

Their custom version of Marlin behaves perfectly with the button so clearly the issue is a setup one.

I have tried many permutations but the best I can get is that the printer will turn in when I press the momentary switch but turns off about a second after I let go.  If I keep it pushed in, the power stays on, leading me to believe that a Marlin setting keeps pulling the PS_ON PIN In the opposite direction.

I’ve tried 1.19 but same issue.  I searched the code for POWER_HOLD, SUICIDE, HAS_POWER_SWITCH, and so on in an effort to understand why this is happening.

I even asked Creatbot for relevant info but their silence is deafening.

To be clear, the Creatbot runs a Rumba based board that uses a a momentary switch to trigger a 24v SSR and ground the PS_ON switch to turn on the printer. The problem is that the PS_ON pin keeps being reset by some routine or commands being sent, maybe from a missasigned pin.

So far all the pins identified are identical to the rumba except for the PS_ON which is on pin 40 instead of 45.  Again, when I do not assign pin 40, it does not show as being used by anything else.

One odd thing I noticed is that the CASE and PS_ON Pin are assigned the same number in original Rumba pins file.

I tried both variations and no luck.  I did trace the PS_ON pin using a multimeter to confirm it and when defined it does respond to M81 so I am sure about it.

The one thing I am not able to I define are the MISO, MOSI, etc pins to validate they are not somehow implicated.  I tried to UNDEF them in the pins files to make sure they were not double assigned but had no luck.  Maybe someone can point me in the right direction although I sincerely doubt they are an issue.

The most likely scenario is that there is some setup that mistakenly sends a command to M40 to pull it in the opposite direction of where it is.  I tried using the DEFAULT_ON Setting as well.

Please excuse my monumentally long post, this is not my printer.  My wife’s company bought it on recommendation and I need to return it ASAP before they look at her sideways.

MrAlvin, 2018-10-13:
As far as I can make out, then Marlin 1.1.9 does not have any code implemented that allows the use of a push button to manage the power system. 

Do you have access to any source code for the Printer? 

Is it possible to turn on the power by issuing command M80 ? 

What happens if you hold the power-on-push-button in, for 5-10 seconds? 

If the power-on-push-button is under software control, then the control board must get power from some other source.  And then command M80 ought to work. 

If the power-on-push-button is however in parallel with the PS_ON pin, then it is very likely that the power-on-push-button needs to be held, until the firmware has booted, and have set the PS_ON pin to take over holding the ON state.

ModMike, 2018-10-13:
I can’t use M80 to turn on the printer because the serial port has no power when it’s off.

I have no issue turning the printer on with the button.  I just can’t turn it off.  If I switch to power supply 2 the inverse is true.

MrAlvin, 2018-10-13:
If the button is parallel with the PS-ON pin, then the button can NOT be used to turn the printer off. 

"Their custom version of Marlin behaves perfectly with the button so clearly the issue is a setup one."
This logic is based on the assumption that their version of Marlin is  only based on modifying settings. If they have added (or modified) the program code, then it will not be possible to repeat the functionality, no matter how much one tries to adjust settings. 

The printer has two power supplies?

ModMike, 2018-10-14:
In a sense yes.  The power button switches the relay that supplies 24V to the controller board.  It was a common setup with Rumba controllers about 2 years ago.

When the power is off, there is still 24V supplied to the switch where it connects to the controller.  I imagine it’s a bypass because the power supply itself is non-atx.

MrAlvin, 2018-10-14:
Sorry, but your descriptions does not compute for me. 

So far I understand that you can turn the printer on, by pushing the On-button. And I also understand that the On-button is connected in parallel with the PS_ON pin. 
This part makes electrical sense to me. And it makes firmware sense to me. 
As the PS_ON pin takes over holding the ON state of the printer after the ON-button has been pushed. 

That the printer can be turned Off issuing command M81 also makes sense to me, as the firmware has program code that supports this action. 

To be able to turn off the printer, by again pressing the ON-button, makes no electrical sense nor firmware sense to me. 
The PS_ON pin is not an input pin, so it would not sense any change. 
Nor is there actually any electrical difference between the ON-button being pushed, when the PS_ON pin holding the power supply in the ON state. 
Nor can I find any program code in Marlin that would support this functionality. 

So if this functionality is possible (with the On-button to be usable as both an ON and OFF button), under  the firmware, that we have no source code access to, then they must clearly have made program code to be able to do this, and they clearly must also use another MCU pin as the input/sense pin to register when the OFF action is needed to happen.

ModMike, 2018-10-14:
That might explain why they would have reassigned the Rumba PS_ON pin from 45 to 40.  Now I’m starting to wonder if pin 45 needs to be assigned to. Any ideas?

ModMike, 2018-10-14:
@MrAlvin I think I found the pin.  I followed the traces and confirmed the path with my multi-tester.  The pin that the switch uses to toggle is Pin 41.  I made up a rough schematic and all other connections are directly related to switching the relay.

So it would seem the there is no code for P41 to trigger M81 off when pulled low (or high).

ModMike, 2018-10-14:
Issue resolved:

//#define LED_PIN            13
#define PS_ON_PIN          40
#define KILL_PIN           41
//#define CASE_LIGHT_PIN     40 

Need to figure out if CASE_LIGHT_PIN  is even needed.

closed 2018-10-14 by ModMike

thinkyhead, 2018-10-16:
I'm very glad you resolved it. Thanks for your feedback and your continued support. Cheers.