MarlinFirmware/Marlin #12078: Compiling error - 2.x bugfix

truglodite, 2018-10-12:
### Description
My configs haven't had problems compiling a month ago, but now there seems to be a problem.

### Steps to Reproduce

1. Compile the latest marlin 2 source using atom+pio and my config files and pins_ramps.h.

### Expected behavior

Compile successfully as usual :)

### What actually happens...

Compile fails, and terminal shows these lines just after the "....elf" line, or just before the summary lines:

SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_9'
.pioenvs/megaatmega2560/src/src/module/endstops.cpp.o (symbol from plugin):(.text+0x0): first defined here
SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_11'
.pioenvs/megaatmega2560/src/src/module/endstops.cpp.o (symbol from plugin):(.text+0x0): first defined here

SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_10'
.pioenvs/megaatmega2560/src/src/module/endstops.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\megaatmega2560\firmware.elf] Error 1
 [ERROR] Took 40.53 seconds

#### Additional Information

I can successfully compile the default source, but when I use my config and pins files it fails. My configuration is a bit atypical due to my ramps having a burned out mosfet and having to move some pins around to compensate. I'm using tmc2130's and a reprap discount smart lcd, so I also had to modify pins_ramps for that. I assure you my config has compiled fine with bugfix 2.x in the past... not sure why the problems now?

configuration.h, configuration_adv.h, and pins_ramps.h attached...
[configFiles.zip](https://github.com/MarlinFirmware/Marlin/files/2474851/configFiles.zip)

It may help that I carefully note any changes to the config files at the top of my configuration.h. I also habitually append "//trug" to all my edited lines.
.

p3p, 2018-10-12:
This is caused by the Software Serial implementation and Endstop Interrupts feature conflicting with each other, I think the only solution is to disable Endstop Interrupts, (as I assume you need the software serial for the tmc2130's)

@teemuatlut may be able to help further.

truglodite, 2018-10-13:
I do have endstop interrupts enabled, but tmc2130's do not use soft serial (2208's do). That was one of my preliminary theories... something got changed with the tmc code and now it is enabling soft serial even though it isn't used/needed.

teemuatlut, 2018-10-13:
It does not matter which driver is used.
The new TMCStepper supports many of the current TMC drivers and depends thus depends on the `SoftwareSerial` library. Any time this header is included it creates these interrupt vectors to be used by the library. That's where the conflict comes and there's no way for external code (TMCStepper) to conditionally include libraries based on internal Marlin configuration settings.

GhostlyCrowd, 2018-10-17:
So the only solution for this is to disable endstop interrupts?

truglodite, 2018-10-17:
teemaultut's post would suggest that (when hardware endstops are essential to a rig with tmc2130's) it should be possible to hack Marlin so soft serial is not included. I am not around my PC to try it out, but searching for '#include SoftwareSerial.h' should be enough to insert the required //'s to prevent the softserial vector problems.

truglodite, 2018-10-17:
A better description for this issue might be "software endstops unecessarily enabled with tmc2130 drivers". So I'm closing this issue. Whoever is further interested in pursuing this should start a new issue report.

closed 2018-10-17 by truglodite