MarlinFirmware/Marlin #12087: Thermal Runaway, system stopped! Heater_ID: bed

REVIEWGAMINGMAN, 2018-10-14:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2018-10-14:
your issues is a safety Features.

TheSFReader, 2018-10-14:
And apart from the title, it seems you didn't in the least try and find out how to report properly the problem you have...

psavva, 2018-10-14:
@REVIEWGAMINGMAN Please can you add the necessary information.

The most important being both the Configuration.h and Configuration_adv.h files.
Given the title, you may either have a hardware issue, or your config files need little adjustments.

REVIEWGAMINGMAN, 2018-10-15:
no everything I have is new parts

REVIEWGAMINGMAN, 2018-10-15:
any idea how to prevent this with a 12-hour print

REVIEWGAMINGMAN, 2018-10-15:
no I did do some research since I was testing my printer for bags

REVIEWGAMINGMAN, 2018-10-15:
the good news is the printer printed for 4 hours with no problems

Grogyan, 2018-10-16:
@REVIEWGAMINGMAN you need to provide a LOT more information in your first post, there is nothing there for us to help you

REVIEWGAMINGMAN, 2018-10-16:
ok

toome123, 2018-10-17:
May be is hardware problem, or config issue..

- You can tune these values in Configuration_abv.h
THERMAL_PROTECTION_PERIOD, 
THERMAL_PROTECTION_HYSTERESIS
- Make sure all connection of the termistors is ok.
- Make sure your power supply is good enough. 
- View what is recommend configurations for your printer...