MarlinFirmware/Marlin #12091: Anet A6 Incorrect Home Position

TehEmoGurl, 2018-10-14:
I just updated my firmware for the very first time and added the E3Dv6 hotend to my Anet A6 printer (running the stock Anet v1.0 main board).

Everything seems to be working fine (Which i am surprised by) except for one thing...

When i use the autohome feature from the menu on the LCD, the X-Axis correctly moves all the way to the Left until it hits its endstop and then the Y-Axis does the same. After this the Z-Axis moves up slightly and the X & Y move until the nozzle is in the center(ish) of the heat bed. Then the Z-Axis slowly lowers until it hit's the endstop switch and then stays there in the center(ish) of the bed.

Originally it would home to the front left corner stopping after all Axis had hit their end stops.

Could anyone help on what settings i need to change in my Configuration.h to set this back to how it originally homed please?

Also, it seems my nozzle is maybe 10-15mm short of the right side of the X axis when moving with the position options the form LCD menu. Once it hits about 235 it wont go any further and can only go back?

Note: i used the Anet A6 default configuration that comes with the latest marlin firmware package download as a base config before changing the settings for my E3Dv6.

Thanks in advance to anyone who can assist

--Jess--

Configuration.h attached below:
[Config 00.zip](https://github.com/MarlinFirmware/Marlin/files/2476493/Config.00.zip)

Ludy87, 2018-10-14:
Disable 

    #define Z_SAFE_HOMING

TehEmoGurl, 2018-10-14:
> Disable
> 
> ```
> #define Z_SAFE_HOMING
> ```

Thank you for the quick response, that fixed the homing issue and is not homing in the correct place.

I now see on the LCD that it is marking the current co-ordinates as: "X-3 Y -5 Z 0"

Is this normal for the latest Marlin version or do i need to fix something?

Also, any idea how to make the nozzle move all the way to the right size of the print bed on the X axis again so it doesn't fall short of the actual build area?

Thanks again

--Jess--

TehEmoGurl, 2018-10-14:
I just rechecked and the X-Axis stops moving at +222 every time and always homes to -3.
The Y-Axis has the full range it should to move but still homes to -5.
Z-Axis seems to be perfect though.

Ludy87, 2018-10-14:
#define X_MIN_POS -3
    #define Y_MIN_POS -5

Go through the configuration file step by step.

TehEmoGurl, 2018-10-14:
> ```
> #define X_MIN_POS -3
> #define Y_MIN_POS -5
> ```
> Go through the configuration file step by step.

I did but there is so much xD

I assume i just set those to 0, or do i comment the lines out?

Will this also allow the nozzle then to move all the way to the right of the bed again?

On more thing, i Re-Enabled the buzzer but it doesn't seem to have activated. i uncommented the "#define buzzer" line. is there something more i have to do to get the buzzer back?

Note: When i re-flash the firmware it no longer reboots the printer like it did the very first time. Yet the Z-axis settings you had me change did work. Do i need to do a manual reboot for some things?

Thanks again, hugely appreciate the help! :)

--Jess--

TehEmoGurl, 2018-10-14:
nevermind it is rebooting, and the Axis are now all homed correctly thank you

fixed the buzzer by changing that line to define buzzer and not the default of speaker. though it doesn't seem to beep on startup :(

it's enough for me to try and start a test print now though so thanks again, have a great day! :)

TehEmoGurl, 2018-10-14:
Sooooooooo i was about to do a test print when i remembered i needed to calibrate the extruder... only to find that it isn't extruding at all since the firmware update. I have tried extruding via the LCD control panel and via OctoPrint and the extruder motor isn't responding at all.

Any ideas? :)

TehEmoGurl, 2018-10-14:
Figured it out. the mintemp for extrude safety feature was stopping me from extruding without first heating the nozzle :)

Ludy87, 2018-10-14:
Great and now close the issues