MarlinFirmware/Marlin #12092: Anycubic kossel sample config not working / Unable to calibrate delta

MrDiba, 2018-10-14:
### Description

When trying to calibrate the Anycubic kossel with the function "Auto Calibration" or "probe Z-offset" the kossel will home itself, the nozzle (with Z-min probe) will move down 4cm and tries to probe mid-air. After that the kossel will home itself and shows a "Probing failed" with a Z height of -34.22.

### Steps to Reproduce

1. Edit the configuration.h file according to the printer, in my case ANYCUBIC_PROBE_VERSION 2 and ANYCUBIC_KOSSEL_ENABLE_BED 1.
2. Upload the Marlin firmware using the Arduino IDE.
3. Install the auto calibration sensor
4. Go to Prepare -> Delta Calibration -> Auto Calibration
5. Result: mid-air probing and a Z-coordinate of -32.24

**Expected behavior:** I expect the nozzle with probe to come down and calibrate the machine by itself.

**Actual behavior:** Calibration happens mid air and fails, 

#### Additional Information

This issue is found in the main 1.1.x branch and the bugfix 1.1.x branch.

b0ssman, 2018-10-15:
see https://github.com/MarlinFirmware/Marlin/pull/12083

you prob need to invert your z min

MrDiba, 2018-10-17:
@b0ssman it was indeed a problem of the inverting z min, thank you for pointing out!

closed 2018-10-17 by MrDiba