MarlinFirmware/Marlin #12097: BLTouch clone deploys for G28 Z0 but signal is not honored by Marlin?

PeterGrace, 2018-10-15:
I'm trying to setup a Geetech BLTouch clone on my i3 clone with Marlin 1.1.9.  With the below changes from 1.1.9-release, M401/M402 commands work properly (deploy/stow) and when a G28Z0 is issued, the probe pole does deploy.  However, when the probe comes in contact with the bed, the gantry does not stop its descent and the extruder crashes into the bed.

I'm curious if I've got anything wrong in the below configuration or what other things I might be able to test.  Meanwhile, I'm going to try putting the endstop signal on the oscilloscope and see if I can get the pulse when it is triggered.

Thanks!

```
+#define BAUDRATE 115200
+#define CUSTOM_MACHINE_NAME "i3PRO"
+#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
+#define TEMP_SENSOR_BED 5
+#define HEATER_0_MAXTEMP 325
+#define EXTRUDE_MINTEMP 160
+#define EXTRUDE_MAXLENGTH 500
+//#define USE_XMIN_PLUG
+#define USE_XMAX_PLUG
+  #define ENDSTOPPULLUP_XMIN
+  #define ENDSTOPPULLUP_YMIN
+#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
+#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
+#define X_DRIVER_TYPE  DRV8825
+#define Y_DRIVER_TYPE  DRV8825
+#define Z_DRIVER_TYPE  DRV8825
+#define E0_DRIVER_TYPE DRV8825
+#define ENDSTOP_INTERRUPTS_FEATURE
+#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 123 }
+#define DEFAULT_MAX_FEEDRATE          { 3000, 3000, 150, 25 }
+#define DEFAULT_ACCELERATION          600    // X, Y, Z and E acceleration for printing moves
+#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
+#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
+#define DEFAULT_XJERK                  7.0
+#define DEFAULT_YJERK                  7.0
+#define DEFAULT_ZJERK                  0.4
+#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
+#define Z_SERVO_ANGLES {10,90}  // Z Servo Deploy and Stow angles
+  #define BLTOUCH_DELAY 50   // (ms) Enable and increase if needed
  *      O-- FRONT --+
-#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
-#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
+#define X_PROBE_OFFSET_FROM_EXTRUDER -45  // X offset: -left  +right  [of the nozzle]
+#define Y_PROBE_OFFSET_FROM_EXTRUDER -11  // Y offset: -front +behind [the nozzle]
 #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
+#define MIN_PROBE_EDGE 2
+#define XY_PROBE_SPEED 12000
+#define MULTIPLE_PROBING 2
+#define Z_PROBE_OFFSET_RANGE_MIN -5
+#define Z_PROBE_OFFSET_RANGE_MAX 5
+#define Z_MIN_PROBE_REPEATABILITY_TEST
+#define INVERT_E0_DIR true
+#define X_HOME_DIR 1
+#define AUTO_BED_LEVELING_BILINEAR
+  #define GRID_MAX_POINTS_X 5
+  #define LEFT_PROBE_BED_POSITION 44
+  #define RIGHT_PROBE_BED_POSITION 150
+  #define FRONT_PROBE_BED_POSITION 3
+  #define BACK_PROBE_BED_POSITION 150
+    #define EXTRAPOLATE_BEYOND_GRID
+    #define ABL_BILINEAR_SUBDIVISION
+      #define BILINEAR_SUBDIVISIONS 5
+#define Z_SAFE_HOMING
+#define HOMING_FEEDRATE_Z  (2*60)
+#define EEPROM_SETTINGS // Enable for M500 and M501 commands
+#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
+//#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.
+#define PREHEAT_1_TEMP_HOTEND 200
+#define ENCODER_PULSES_PER_STEP 2
+#define ENCODER_STEPS_PER_MENU_ITEM 4
+#define SPEAKER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
+#define SERVO_DELAY { 50 }
+  #define WATCH_TEMP_PERIOD 40                // Seconds
+  #define THERMAL_PROTECTION_BED_PERIOD 40    // Seconds
+#define MINIMUM_STEPPER_PULSE 2
+#define ADVANCED_OK
```

PeterGrace, 2018-10-15:
I hooked it up to the scope and did not see any signal when the plunger was toggled.  I'm wondering if this is a bum unit.

PeterGrace, 2018-10-15:
I did a diff; if default marlin 1.1.9 has the setting, it wouldn't show up there.  I do have both of those set properly.  I can see now why they say to put a zip file of the header files, my mistake.  Since I wasn't able to get a signal from the touch probe, I've ordered a replacement to double check that it wasn't just a DOA sensor.  I'll report back if that was the issue.

TheSFReader, 2018-10-15:
PeterGrace, are you sure the pin you're testing is pulled up/down correctly (can depend/be specified on the configuration). If not, the "signal" can be lost in floating state.

PeterGrace, 2018-10-15:
@TheSFReader I was testing the sensor by attaching the ground to the ground wire and the signal to the probe wire.  That should bring the scope and the touch sensor into the same ground potential, should it not?

TheSFReader, 2018-10-15:
Far from being a master of electronics, but I guess it depends on what the signal wire is ALSO connected to. 

I don't know if the BL-Touch internally pulls up or down the signal, or if it depends on the control board to do that. 

And I don't see any ENDSTOP_PULLUP_ZMIN in your configuration. 

(You Can STILL push the whole config file yes ?)

PeterGrace, 2018-10-15:
Sorry, here's the configs.

[i3pro-config.zip](https://github.com/MarlinFirmware/Marlin/files/2480634/i3pro-config.zip)

Phr3d13, 2018-10-16:
Not if you're setting the value via M851 after flashing.

PeterGrace, 2018-10-16:
Yep, I set the offset via M851.  and @cdedwards thanks for the BLTOUCH suggestion.  I have tried it with BLTOUCH enabled, and without.  If BLTOUCH is enabled, should this section also be enabled?
```
#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {10,90}  // Z Servo Deploy and Stow angles
```

Phr3d13, 2018-10-16:
I use a BLTouch on one of my machines, and I don't have those enabled.

PeterGrace, 2018-10-19:
After receiving a replacement probe, the issue is with the original probe.  The new probe works as expected.

closed 2018-10-19 by PeterGrace