MarlinFirmware/Marlin #12099: Babystepping not working

rb1205, 2018-10-15:
### Description

On current bugfix-1.1.x (last commit 5e0dffd) i'm having big issues with babystepping. It's like no matter what direction I rotate the dial, the Z axis moves in one direction. Sometimes it's up, sometimes it's down, but the direction is not related to the way I move the encoder.
The value in the LCD changes correctly based on how I move the encoder (up or down), so it's not an issue with the encoder.

I have UBL compiled and active

Tried both with BABYSTEP_ZPROBE_OFFSET enabled or disabled, same behavior.

### Steps to Reproduce

1. Home the axis
2. Move the Z axis next to the printing bed, eg. G1 Z1
3. Initiate a slow XY movement, eg G1 F20 X20
4. Go to the babystepping page either via double click or by navigating the menu
5. Rotate the encoder clockwise, notice how the Z axis moves
6. Rotate the encoder counter-clockwise, notice how the Z axis moves

**Expected behavior:** [What you expect to happen]
The Z axis moves both up and down based on which way I turn the encoder

**Actual behavior:** [What actually happens]
The Z axis moves the same direction regardless of which way I turn the encoder

#### Additional Information

[Configuration ZIP file](https://www.dropbox.com/s/dxiglw1axyp0dt2/Configuration.zip?dl=0)

thinkyhead, 2018-10-16:
Try increasing your `MINIMUM_STEPPER_PULSE` value.

Fr33man85, 2019-12-29:
What is the conclusion, what solved the problem?

changes at MINIMUM_STEPPER_PULSE 

or 

changing the driver?

rb1205, 2019-12-30:
Neither. I'm using industrial grade drivers that have strict requirements in term of direction reversal timings that can be violated by babystepping coupled with auto bed levelling, no matter how I configure Marlin.

Long story short, babystep only works if the z axis is halted (duh) OR if the babystep movement and the level compensation movements are the same direction. If you try to babystep the "wrong" direction, the Z axis will lose steps and "snap" back. Kinda hard to explain, but with a bit of experience you learn how to make it work correctly.  It's not ideal, but the reliability and power of these drivers make up for that long time. I'm finally done with losing prints due to overheating/underpowered/failing/whatever a4988s & co. threw at me.

I guess that a more configurable PWM timing system on Marlin's end could solve that, but again it's not a big deal right now.