MarlinFirmware/Marlin #12103: [2.0.x] Inconsistent behavior of software endstops with NO_WORKSPACE_OFFSETS vs without

marcio-ao, 2018-10-15:
### Description

When ENABLED(NO_WORKSPACE_OFFSETS), it is not possible to move beyond XY_BED_SIZE unless software endstops are disabled. This is different than what [documentation to M211 states](http://marlinfw.org/docs/gcode/M211.html) as the behavior.

With  DISABLED(NO_WORKSPACE_OFFSETS) it is possible to reach XY_MAX_POS, which is the correct behavior as written in the documentation.

### Steps to Reproduce

1. Set X_MAX_POS to something > X_BED_SIZE
2. Compile without NO_WORKSPACE_OFFSETS
3. Try moving to X_MAX_POS
4. The motion will reach X_MAX_POS
5. Recompile with NO_WORKSPACE_OFFSETS
6. The motion will only proceed up to X_BED_SIZE, then stop

thinkyhead, 2018-10-26:
> Try moving to X_MAX_POS

With `G1` ?

It's an open question whether the software endstops should clip to the bed or to the physical limits. The `clamp_to_software_endstops` function is meant to keep a print from exiting the bed, so that implies that it should constrain to the bed size. But under normal (pre-printing) conditions it's more convenient to have them use the physical bounds.

I'm unable to find the reason why `NO_WORKSPACE_OFFSETS` would have an effect after scanning the code. Maybe you'll have better luck.

boelle, 2019-02-21:
@marcio-ao is this one still "un-solved" ?

marcio-ao, 2019-02-21:
@boelle: Just tried with the latest bugfix-2.0.0 and I am unable to reproduce the problem. Will close this down as fixed.

closed 2019-02-21 by marcio-ao

boelle, 2019-02-21:
yaaa.... one less

allow me to be naive and say that we soon only have one page with real issues  and feature request on the rest :-D

marcio-ao, 2019-02-21:
@boelle: I think you are being too optimistic. I'll bet on two pages!