MarlinFirmware/Marlin #12106: LCD screen hangs up at random

edism, 2018-10-16:
I use latest 1.1.9 firmware.  MKS GEN 1.4 board with full graphics LCD and simplify UI.

Sometimes when starting a print, LCD screen hangs up and is not responding or being updated until print finishes. After printing completes, everything comes back to normal again. 

Any thoughts what could be causing this?

edism, 2018-10-16:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2482442/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2482443/Configuration_adv.txt)

thinkyhead, 2018-10-16:
I have no idea. Maybe there's too much electrical noise in your machine interfering with the LCD connection. I'm going to have to drop a canned reply on you, sorry…

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-10-16 by thinkyhead

AnHardt, 2018-10-16:
For the graphic displays we monitor how long a display update takes, and if there is enough time for this in the planer buffer. If there is not enough time in the buffer (to short blocks or not enough blocks) the display update is delayed until there is enough time to do it.
Try printing longer segments.

edism, 2018-10-16:
I have printed the same part twice. First time everything was ok, LCD was being updated and i was able to tweak flow seetings, second time after starting print the display was frozen and i have witnessed it coming back  to life after print was finished.

I have not changed a thing in my printer for the last  year  and it was working fine before.

AnHardt, 2018-10-16:
Check the hardware. Code has not changed for a while.

thinkyhead, 2018-10-20:
We’ll keep an eye on it. We know there are many using the lightweight UI without issue, so perhaps it requires some magic combination of features, hardware, and planetary alignments. You may find that disabling other features also fixes the issue. Keep troubleshooting, and report if you find anything else interesting,