MarlinFirmware/Marlin #12107: G28 Enstop ignore

DvDriel, 2018-10-16:
Endstops are ignored when using G28 command. I am building a delta with an UltraTronics board (ATSAM3x8E) and Marlin 2.0. Marlin compiles and uploads fine, G119 gives expected results:
x_max: open
y_max: open
z_max: open

G1 also works like expected, when the endstops are hit the motors stop. When I use G28 however the endstops are ignored and the motors keep turning.

I don't expect this to be a bug in Marlin, but I tried everything and can't seem to figure out what the problem is.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2482201/Configuration.zip)

thinkyhead, 2018-10-16:
We know the endstops code works when the endstops work because we use them every day. So something is certainly wrong with your endstops or your endstops configuration.

DvDriel, 2018-10-16:
I agree, but I was wondering if anybody knows what could possibly be the issue. I have been struggling with it for a couple of months already. The endstops work perfectly with M119 and G1, but only G28 causes problems.

thinkyhead, 2018-10-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-10-16 by thinkyhead

JasonKarlaile, 2018-10-16:
It reported z min as "triggered" while the extruder was fully down and "open" everywhere else. The endstops seem to be working with the software on some level, so I'm not sure where to go from here. Given that I wanted to keep things simple and stick with MatterControl for now, do you think my best option would still be to switch to Repetier or something similar? I checked  here https://goo.gl/VS81cM for the best solution