MarlinFirmware/Marlin #12109: Marlin 2, SERIAL_PORT_2 compile error...

axl12ap, 2018-10-16:
Hi,
I've the Arduino2 with RuRamps 1.3 shield, and I'd like to use the Serial2 port connected to my ESP8266 with ESP3D-WebUI webserver.

That's what I want:
- Serial 0 : Programming port / USB
- Serial 2 : ESP8266 ESP3D / WiFi

But if I enable it the option on the Marlin Configurator (and I select "2") when I compile the firmware give me back this error:

SanityCheck.h:369:4: error: #error "SERIAL_PORT_2 is not supported for your MOTHERBOARD. Disable it to continue."

Any ideas?

Ludy87, 2018-10-16:
Post your Configuration.h line

    #define SERIAL_PORT
    #define SERIAL_PORT_2

axl12ap, 2018-10-17:
Sorry, can be close due for the update of firmware with the GitHub version here... my local version was not updated!

Now I'm able to compile with SERIAL_PORT_2 also!
Thanks

closed 2018-10-17 by axl12ap