MarlinFirmware/Marlin #12110: Markforged kinematics

someoneprivate, 2018-10-16:
Hello there, 
Im not sure this is the right place for this, but regardless, Im wondering how and is it possible to incorporate some rudimentary kinematics that aren't "supported" by Marlin natively. As such, I will say that I am not really familiar with Marlins source code and would greatly appreciate any and all tips regarding my implementation.
I goes as such, I have double Y axis motors that go linearly, Z being similar
with X axis I have implemented something similar to a markforges design.
[image]

TheSFReader, 2018-10-16:
Looks like the Kinematics you need are an hybrid between cartesian and Core_XY, so I think the best way to evaluate the code changes needed to implement this Kinematic would be to check out all the places where CORE is used . 

That would be mainly the planner, endstops, and temperature (babystepping).

someoneprivate, 2018-10-17:
Is there any documentation/knowhow regarding such a thing?

TheSFReader, 2018-10-17:
I don't think so, no. The knowhow is in the code, and as I wrote eralier the CORE related codebase would be a great starting point/guideline for your needs.

someoneprivate, 2018-10-17:
Anyway, thank you for your help, thought, I think I found my solution in Repetier Firmware

thinkyhead, 2018-10-17:
Essentially, Markforged is just like CoreXY except that only one stepper is treated differently, rather than two. Marlin uses `IS_CORE` to delimit code that does things differently for Core kinematics, and then it uses `CORE_IS_XY`, `CORE_IS_XZ`, and `CORE_IS_YZ` to delimit each subtype. To get Markforged working we would just need to add `COREMF` to configurations, then modify the code within the `CORE_IS_XY` blocks to move the A stepper in proportion to Y, while still moving the B stepper in proportion to X minus Y, as we already do for `COREXY`.

To accommodate possible variations of Markforged, the same special case would be added under `CORE_IS_XZ`, and `CORE_IS_YZ`.

AnHardt, 2018-10-17:
> variants YX, ZX, and ZY

Do they really exist or can the mirrors and rotations all be handled by changing (A and B) and/or (changing axis directions)?
But like with the Cores no user will get that. Like the are always confused when the bed moves.

TheSFReader, 2018-10-17:
An other way would be to "unify"  the equations between all cartesian + core + markforge configurations as 

deltaStepperA = factorXA * deltaX + factorYA * deltaY
deltaStepperB = factorXB * deltaX + factorYB * deltaY 
(and equivalent for XZ... etc)
and just use defines with different factors for the different core/Markforge  configurations.
(would the compiler be able to compile out zeroed out factors ?)

But still need to manage the endstops properly, which can also prove a challenge...

someoneprivate, 2018-10-18:
Hmm, It would be possible, but I now see why no one has attempted such a design. + If I was to do it as such, Im pretty sure, I would need to rewrite the whole homing process, not to say somehow implement dual Y axis endstops.   
In any case, Im far to incompetent to do either of them, so I think I'll just change it to a thoroughbred cartesian machine in the design.

marcio-ao, 2018-11-19:
> An other way would be to "unify" the equations between all cartesian + core + markforge configurations as
> 
> ```
> deltaStepperA = factorXA * deltaX + factorYA * deltaY
> deltaStepperB = factorXB * deltaX + factorYB * deltaY
> ```

Oh no, I am getting flashbacks to my linear algebra classes... :O

0xmarcelx0, 2019-09-26:
Hello guys, I converted a Tronxy-X5sa to this Markforged design (but with two carriages).

[image]

Unfortunately I didn't know that this design doesn't exist in the marlin firmware yet.  Do I have a chance to make this design work with Marlin?

I had chosen this design to shorten the belt length and to use two carriages without having to move the stepper motors during printing. Also, no belts cross each other.

Please help me.

Best regards from the middle of Germany :-)

Translated with www.DeepL.com/Translator

someoneprivate, 2019-09-28:
Hello there, (and here I was thinking this thread was long dead...)

I guess its due time I made an update, basically I ravaged the source code of a common XY machine, flipped some of the kinematics around and in the end wasn't to difficult, and bonus, dual  endstops were not required... thank fuck, anyhow, I like what you've decided to do here, but as I see it wasn't to be, Marlin in the end really isn't as flexible to allow for nisch designs as much as people would like...
But I would like to divert your attention to a much simpler solution then what I went through which is using repetier, heres there online config, you can achive much of what you want in there https://www.repetier.com/firmware/v100/index.php

Greetings from Slovenia and as such please excuse my english :)

someoneprivate, 2019-09-28:
> Hello guys, I converted a Tronxy-X5sa to this Markforged design (but with two carriages).
>
> [image]
>
> Unfortunately I didn't know that this design doesn't exist in the marlin firmware yet. Do I have a [...]

Hello there, (and here I was thinking this thread was long dead...)

I guess its due time I made an update, basically I ravaged the source code of a common XY machine, flipped some of the kinematics around and in the end wasn't to difficult, and bonus, dual endstops were not required... thank fuck, anyhow, I like what you've decided to do here, but as I see it wasn't to be, Marlin in the end really isn't as flexible to allow for nisch designs as much as people would like...
But I would like to divert your attention to a much simpler solution then what I went through which is using repetier, heres there online config, you can achive much of what you want in there https://www.repetier.com/firmware/v100/index.php

Greetings from Slovenia and as such please excuse my english :)

hussainsail2002, 2019-12-07:
Has anyone figured out how to do this in marlin, it will be interesting to try this out.

hzmeister, 2019-12-22:
I'd also REALLY like a solution to get markforged kinematics on marlin 2.0.
Reprap firmware(for duet boards) recently had it implemented and it's easy to setup with a single gcode command.

Koenert, 2020-06-18:
> Hello guys, I converted a Tronxy-X5sa to this Markforged design (but with two carriages).
> 
> [image]
> 
> Unfortunately I didn't know that this design doesn't exist in the marlin firmware yet. Do I have a chance to make this design work with Marlin?

Hi everyone,

Are there any updates on this? 
I'm considering using the markforged kinematics because the motors for the carriages then don't have to sit on the gantry anymore. This way the gantry will have less weight so the print speeds can go up a bit. This is needed because the printarea for my newly designed printer will be 400x400x600mm. 
It would be nice to be able to stick with marlin because I'm used to it.

Kind regards from the Netherlands :)

thinkyhead, 2020-06-19:
I'll take another crack at Markforged, but no guarantee I'll get it right. Also, we do not even have support for Dual X Carriage in our standard CoreXY handling now, so that would also need to be implemented.

Koenert, 2020-06-20:
Hi, 
Thank you for looking into it. I know it will be a big job but probably more people would be able to stick with Marlin when it offers compatibility with more complicated motion systems. 
I like Marlin due to the fact it's pretty easy to modify for your specific needs and that it works. Thanks a lot for all your effort!

AnHardt, 2020-06-20:
A Markforge Dual X Carriage system does need only 3 logical axes. On of these would be able to drive 2 steppers. Both off the y-steppers always have the same position and movement.
Because at a Core-xy Dual X Carriage system the y-stepper can not be common it needs a fourth logical axis

Koenert, 2020-06-20:
That's true indeed, the Y-steppers will be in sync (like 2 steppers for z-axis) or you are able to use only one y-stepper by connecting the two sides. I want to configure it like shown in the picture below, because that way I only need one y-stepper. 

[image]

qwerty208, 2020-06-24:
Hello. I designed and built a printer with this kinematics, but I still could not fix the Marlin formulas for work. It turned out my experience and knowledge of Marlin is not enough. Would love to see the Markforged kinematics in Marlin.
Best regards from Russia. (Google Translate)
[image]
[image]

qwerty208, 2020-06-29:
Hello. I managed to get Marlin2 working with Markforged kinematics. Fixed lines **IS_CORE** and **CORE_IS_XY** in the files planner.cpp and stepper.cpp
I ask you to pay attention to the line 1796 planner.cpp, this is how the limit switch at maximum Y (**Y_HOME_DIR 1**) worked. Having printed out 3 details, I did not identify any malfunctions, but we still need tests for more confidence.
I post the status of the files at the moment.
Google Translate.
[image]

Fabiandelete, 2020-07-30:
Hi Everyone,

I'm in the same situation as @Koenert , currently building a new large scale printer. My plan from the beginning was to use Marlin on a ski 1.1 pro. And Ultimaker style kinematics, only doubled up for idex.

I just found this thread and it's inspiring. The reason I chose the Ultimaker style kinematics is so I can have static steppers on the idex. But it also involves a lot of lineair guides/bearings. All of which add to complexity and possible jams.

The Makerforged kinematics look extremely promising and I would love for it to be usable with Marlin.

Keep up the good work!

sjasonsmith, 2020-09-11:
This was merged a few days ago. I assume this can now be closed?

sjasonsmith, 2020-11-10:
This is complete. Closing.

closed 2020-11-10 by sjasonsmith

RootCNC, 2021-01-11:
Hi everyone - I saw this thread and thought it was a great idea! 

Has either the Markforged or HaqXY kinematics been implemented yet? I've currently been building this printer (its still a working progress) 
[image]

I've just started playing around with my new SKR PRO V1.2 to drive it. However, Markforged kinematics doesnt support dual X carriages.

Can someone point me in the direction to implemented either configuration with Dual X carriages?

PS it Pete from Root CNC :)

Thanks in Advance

Fabiandelete, 2021-01-13:
@RootCNC That's an awesome looking printer! Sadly, as you mentioned, markforged doesn't support idex. I know that reprap does support it and that the skr pro should be able to run reprap with some additional hardware.

RootCNC, 2021-01-13:
Well bugger! I've always used marlin. I don't believe Reprap firmware is ported to the STM32 boards yet. Well I didn't think this was going to be an issue. That's a shame