MarlinFirmware/Marlin #12113: Progress bar not showing status when less than 2-3%

mmoraes80, 2018-10-16:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

mmoraes80, 2018-10-16:
Just start a print job. The progress bar will stay empty and no % number will be shown until progress >= 2%.

thinkyhead, 2018-10-26:
This is by design, to preserve the status message for longer, but if you want to change it, just find the following text in the LCD code and change it to suit your preference:
```
progress_bar_percent > 2
```

closed 2018-10-26 by thinkyhead