MarlinFirmware/Marlin #12114: Error compiling TMC2130

sergiofagundes, 2018-10-16:
I'm trying to compile for TMC2130 and got the error bellow in Arduino 1.9.0beta:

===========
In file included from sketch\src\module/stepper.h:224:0,

                 from sketch\src\Marlin.cpp:36:

sketch\src\module/stepper_indirection.h:61:26: fatal error: TMCStepper.h: No such file or directory

compilation terminated.

exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560
==================

I have the library installed in Arduino:
TMC2130Stepper by teemuatlut 2.5.0

MikeLud, 2018-10-16:
@sergiofagundes 
Below is a link to TMCStepper library.
[https://github.com/teemuatlut/TMCStepper](url)

sergiofagundes, 2018-10-16:
Got it to compile. Was using version 2.5.0, but the link you sent works v0.2.2 perfectly. Thank you.

MikeLud, 2018-10-17:
@sergiofagundes 
TMCStepper V0.2.2 is a newer library replacing TMC2130Stepper V2.5.0

closed 2018-10-26 by thinkyhead

Vanuan, 2018-12-07:
Doesn't work for me. Which version of marlin is it compatible with?
```
In file included from sketch/stepper.h:194:0,
                 from sketch/Marlin_main.cpp:261:
stepper_indirection.h:60:30: error: TMC2130Stepper.h: No such file or directory
compilation terminated.
exit status 1
TMC2130Stepper.h: No such file or directory
```

```
#if HAS_DRIVER(TMC2130)
  #include <TMC2130Stepper.h>
  void tmc2130_init_to_defaults();
#endif
```

thinkyhead, 2018-12-07:
>  Which version of marlin is it compatible with?

You'll need to migrate to `bugfix-2.0.x` to get the latest TMC-oriented improvements. Marlin 1.1.9 also uses the TMCStepper library in place of the old libraries.

Vanuan, 2018-12-08:
The above error is from latest 1.1.x git. So is it expected to work?

thinkyhead, 2018-12-08:
It should work if you have `TMC2130Stepper` library 2.2.1 or newer installed in your Arduino libraries.