MarlinFirmware/Marlin #12116: E0 no longer works

eulogix, 2018-10-16:
Made my own fork at commit 3d5be2ea4bac370e05b8acc61a9fba2684e5131e, and changed some stuff in Configuration.h to match the old config I had for 1.x. Board is an MKS S-BASE, printer is a Tevo Tornado.
All worked fine, then merged the changes from 2.0 at commit bf0c809ff2de1cf0dcd750e2adb2760f30edf2f5, and suddenly E0 stopped working. No change to the config.
Tried reverting the merge and it worked again. The motor hisses and holds the position, but doesn't move.
I have TMC external drivers, tested with both the TMC and the onboard DRV, no change.

gloomyandy, 2018-10-16:
Please post your configuration files.

eulogix, 2018-10-16:
Here you go: only mod in pins_MKS_SBASE.h is a servo pin switch because the default one made the bltouch misbehave.
Note that with these configs, Marlin at https://github.com/MarlinFirmware/Marlin/commit/3d5be2ea4bac370e05b8acc61a9fba2684e5131e
runs just fine (printed a cube right now)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2485045/config.zip)

visallard, 2018-10-16:
Same issue, commit https://github.com/MarlinFirmware/Marlin/commit/056561df935f47ecfabc16e2ed777f3754be837c, E0/1 are completely unresponsive.

I've checked the drivers and steppers, and they work on regular XYZ movement. Disabling cold extrusion prevention doesn't change a thing.

I'm using an MKS-SBASE with the onboard drivers with an Anycubic Kossel Linear+.

The config files : 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2485049/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2485050/Configuration_adv.txt)

AletheianAlex, 2018-10-16:
Same issue here on a simple lab rat testbed printer (ramps+rearm+a4988).  2.0 Bugfix from this morning = no extruder.  Reverted to the same commit as above = working fine. Only config changes from stock configuration.h in the marlin folder are:

Printer specific: DEFAULT_AXIS_STEPS_PER_UNIT, bed size, filament diameter, temp sensor, PID
#define TEMP_SENSOR_0 5
#define  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define SDSUPPORT
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB

Just re-confirmed that in a diff tool.  Only other change in any file from default setup are LPC1768 in the *.ini file.

visallard, 2018-10-16:
Note : I'm using HEATER_1_PIN for the print cooling fan

p3p, 2018-10-16:
@eulogix don't have a solution to this right now, there have been a few merges to do with extruding since my big framework PR, but in regard to the misbehaving bltouch was that before or after https://github.com/MarlinFirmware/Marlin/commit/3d5be2ea4bac370e05b8acc61a9fba2684e5131e as I pretty much rewrote the Servo and PWM code.

p3p, 2018-10-16:
Confirmed https://github.com/MarlinFirmware/Marlin/commit/f56968b (#12098) as the first commit to exhibit this behaviour, looking into it now.

p3p, 2018-10-16:
Isolated bug @ [stepper.cpp#L1415-L1417](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.cpp#L1415-L1417)

`HAS_E_STEP` does not exist it needs to be changed to `HAS_E0_STEP` or check whether HAS_E[0-5]_STEP is true instead, (@thinkyhead)

closed 2018-10-18 by thinkyhead