MarlinFirmware/Marlin #12119: Backup/Restore settings to/from SD Card

sergiofagundes, 2018-10-17:
Request for a feature to backup to / restore from SD Card:

- Auto Calibration data
- Auto leveling data

swilkens, 2018-10-17:
We have something along these lines, it just doesn't write it to SD card yet (afaik).

The `M503 S0` command will output the current EEPROM data as a replayable gcode file, meaning the output of `M503 S0` could be save as a gcode file and executed on the printer to restore settings.

So a function to make `M503 S0` dump to SD card might be enough to obtain this. Add it to the LCD menu as well perhaps?

sergiofagundes, 2018-10-17:
This is helpfull, thank you.

thinkyhead, 2018-10-26:
It would be a significant rewrite to add save to SD capability for `M503` because every printed string would need to go into a stack frame buffer. In the future the EEPROM code will get some rewrites, and at that point we'll consider such a possibility.

guitaristtom, 2021-08-12:
Great idea. It would be quite helpful.

brosskgm, 2022-01-09:
I tried this and after I installed the new SKR board and after running the gcode it loaded everything except the leveling mesh back to the eeprom.

Any additional options to get leveling mesh uploaded?

> We have something along these lines, it just doesn't write it to SD card yet (afaik).
> 
> The `M503 S0` command will output the current EEPROM data as replayable gcode, meaning the output of `M503 S0` could be save as a gcode file and executed on the printer to restore settings.
> 
> So a function to make `M503 S0` dump to SD card might be enough to obtain this. Add it to the LCD menu as well perhaps?
> 
> For example, this is the (verbose, so ignore the READ: and SENT: statements) output on my printer:
> 
> ```
> > SENT: M503 S0
> > READ:   G21    ; (mm)
> > READ:   M149 C ; Units in Celsius
> > READ: 
> > READ:   M200 D1.75
> > READ:   M200 D0
> > READ:   M92 X80.00 Y80.00 Z400.00 E93.00
> > READ:   M203 X500.00 Y500.00 Z5.00 E50.00
> > READ:   M201 X500 Y500 Z100 E5000
> > READ:   M204 P500.00 R500.00 T500.00
> > READ:   M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
> > READ:   M206 X0.00 Y0.00 Z0.00
> > READ:   M420 S1 Z0.00
> > READ:   G29 S3 X1 Y1 Z0.00000
> > READ:   G29 S3 X2 Y1 Z0.00000
> > READ:   G29 S3 X3 Y1 Z-0.10000
> > READ:   G29 S3 X1 Y2 Z0.00000
> > READ:   G29 S3 X2 Y2 Z0.00000
> > READ:   G29 S3 X3 Y2 Z-0.10000
> > READ:   G29 S3 X1 Y3 Z-0.17500
> > READ:   G29 S3 X2 Y3 Z-0.17500
> > READ:   G29 S3 X3 Y3 Z-0.15000
> > READ:   M145 S0 H185 B45 F255
> > READ:   M145 S1 H240 B0 F255
> > READ:   M301 P21.73 I1.54 D76.55
> ```