MarlinFirmware/Marlin #12122: Continuous Printer halted. kill() called! when using TMC2130

sergiofagundes, 2018-10-17:
When using latest Marlin 2.0 bugfix I got continuous error reseting printer with a Continuous Printer halted. kill() called!.

Before on bugfix dated 10/14/2018 this didn't occur, but I got another printer halted when I begin to print and get a hotend E0 error.

AnHardt, 2018-10-17:
A kill() has always a reason. Most times you can find it in the log, a few lines above the kill-message.

sergiofagundes, 2018-10-17:
Fixed it, commenting the line in configuration_adv:

//#define STOP_ON_ERROR

When this line is active I got continuous error.

ManuelMcLure, 2018-10-17:
You really want to find out why you're getting errors. All you did with that change is make Marlin ignore the error - given that you were getting an E0 error this could be dangerous (if it's a thermal overrun or similar).

AnHardt, 2018-10-17:
Luckily `STOP_ON_ERROR` has no effect on the heater system (else i'd make a PR to remove it immediately), but with the TMC drivers.

teemuatlut, 2018-10-17:
`STOP_ON_ERROR` was implemented to deal with a case where a TMC driver has shut itself down due to an error like over temperature or short to ground. When the driver protects itself like this, it'll stop responding to step pulses and the firmware will have no way to react to this unless we poll the driver status. If Marlin doesn't know one of the axis has stopped moving, it'll happily keep "printing" a failed part and wasting your filament.
Future improvements are coming that include some sanity checking against invalid responses, fetching data only once per driver and requiring multiple errors before killing a print.

closed 2018-10-26 by thinkyhead