MarlinFirmware/Marlin #12123: [FR] Show remaining print time with M73

annego15, 2018-10-17:
The Slic3r Prusa Edition added a feature to show the remaining print time on the lcd by sending "M73 P{percent} R{remaining time in minutes}". 

M73 already exists in Marlin to set the progress in percentage, but i lacks the R argument for remaining time in minutes.

It would be really nice if you could add this to Marlin so you can see an accurate estimate from the slicer how many minutes until the end instead of the elapsed time.

AnHardt, 2018-10-17:
Use M117 instead. No space left.

thinkyhead, 2018-10-19:
> No space left.

Of course, it could blink between Elapsed time and ETA if space is tight.

boelle, 2019-03-13:
@thinkyhead was this not added recently?

GMagician, 2019-10-17:
This may be closed, I think

closed 2019-10-17 by annego15