MarlinFirmware/Marlin #12126: Missing steps on "recommended" settings speed tuning- Acceleration/Jerk

Yoonutz, 2018-10-17:
### Description

I have built a new cube printer with build volume 350x440x450 mm. And With recomended settings from Tom's video on the internet (https://www.youtube.com/watch?v=7HsIZuj9vOs&t=278s) I have huge missing steps on Y axis (and sometimes on X axis)

### Steps to Reproduce

I have tightened the belts as much as I could in first place. The result is the same.
I have released the belts to a "ok-ish" point. Same result
I have decreased the values as shown below
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z10.00 E20.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X200 Y200 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P20.00 R3000.00 T3000.00

**Expected behavior:** 

High acceleration (M204 ~P9001 ) without losing steps. For further calibrations

**Actual behavior:** 
Went down to  slow acceleration (M204 P20) in order to have no missing steps. I still see remains of missing steps, but they are acceptable now. 
At this slow rate, I dont think it's a mechanical problem

#### Additional Information

Below are some photos of how the calibration looks now, after lowering the acceleration and having ok-ish results. For detailed prints, this speed is OK, but I want to go on higher speeds for bigger prints (given that I have a build volume of 350x440x450mm)
[https://scontent.fotp3-2.fna.fbcdn.net/v/t1.15752-0/p480x480/44115367_1934443493292122_5050499202107310080_n.jpg?_nc_cat=105&oh=397e6eb2bc254a2f336b2e66fd97fe1d&oe=5C3E27A5](url)

[https://scontent.fotp3-2.fna.fbcdn.net/v/t1.15752-9/s2048x2048/44136671_2181296285460679_3276741278670258176_n.jpg?_nc_cat=102&oh=f6be3365d2ee05e153ed8611d2aa7906&oe=5C50C346](url)

[https://scontent.fotp3-2.fna.fbcdn.net/v/t1.15752-9/s2048x2048/44110974_338933906670323_4148680258926673920_n.jpg?_nc_cat=100&oh=c799dea4e9300f1674eebde8e0441103&oe=5C58A2D8](url)

swilkens, 2018-10-17:
Given that you are runnig these drivers fairly heavy without heatsinks, overloading the drivers seems like a plausible cause here.

Which mode are you operating the TMC2130 drivers in? Is this is stealthChop?

You might try compiling Marlin with `#define TMC_DEBUG` in `configuration_adv.h`. After doing that, you can use `M122 S1` to get debug output.

Yoonutz, 2018-10-17:
I have done that before. Below are my results usint just M122:
>>> M122
SENDING:M122
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	800	900	1000	800
RMS current	795	887	994	795
MAX current	1121	1251	1402	1121
Run current	25/31	28/31	17/31	25/31
Hold current	15/31	16/31	10/31	15/31
CS actual		15/31	16/31	10/31	15/31
PWM scale		58	47	39	2
vsense		1=.18	1=.18	0=.325	1=.18
stealthChop	true	true	true	true
msteps		16	16	16	16
tstep		1048575	1048575	1048575	1048575
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		5	5	5	5
blank time	24	24	24	24
hysteresis
-end		2	2	2	2
-start		3	3	3	3
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E0
stallguard
sg_result		0	0	0	0
fsactive
stst		X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x80:0F:00:00
	Y = 0x80:10:00:00
	Z = 0x80:0A:00:00
	E0 = 0x80:0F:00:00

And using M122 with S1 command:

X:57 |0b1000| 	Y:77 |0b0| 	Z:37 |0b1000| 	E0:1 |0b1000|
X:61 |0b1000| 	Y:74 |0b1000| 	Z:38 |0b1000| 	E0:1 |0b1000|
X:58 |0b1000| 	Y:79 |0b1000| 	Z:41 |0b1000| 	E0:1 |0b1000|
X:60 |0b1000| 	Y:76 |0b1000| 	Z:36 |0b1000| 	E0:1 |0b1000|
X:59 |0b1000| 	Y:77 |0b1000| 	Z:37 |0b1000| 	E0:1 |0b1000|
X:56 |0b1000| 	Y:75 |0b1000| 	Z:69 |0b0| 	E0:1 |0b1000|
X:63 |0b1000| 	Y:42 |0b1000| 	Z:70 |0b0| 	E0:1 |0b1000|
X:94 |0b0| 	Y:44 |0b1000| 	Z:62 |0b1000| 	E0:1 |0b1000|
echo:busy: processing
X:109 |0b0| 	Y:47 |0b1000| 	Z:61 |0b1000| 	E0:1 |0b1000|
X:111 |0b0| 	Y:45 |0b1000| 	Z:63 |0b1000| 	E0:1 |0b1000|
X:110 |0b0| 	Y:46 |0b1000| 	Z:64 |0b1000| 	E0:1 |0b1000|
X:109 |0b0| 	Y:43 |0b1000| 	Z:63 |0b1000| 	E0:1 |0b1000|
echo:busy: processing
X:99 |0b0| 	Y:45 |0b1000| 	Z:64 |0b1000| 	E0:1 |0b1000|
X:97 |0b0| 	Y:42 |0b1000| 	Z:41 |0b1000| 	E0:1 |0b1000|
X:93 |0b1000| 	Y:120 |0b0| 	Z:40 |0b1000| 	E0:1 |0b1000|
X:94 |0b1000| 	Y:116 |0b0| 	Z:38 |0b1000| 	E0:1 |0b1000|
echo:busy: processing
X:89 |0b1000| 	Y:124 |0b0| 	Z:37 |0b1000| 	E0:1 |0b1000|
X:98 |0b1000| 	Y:117 |0b0| 	Z:40 |0b1000| 	E0:1 |0b1000|
X:91 |0b1000| 	Y:90 |0b0| 	Z:38 |0b1000| 	E0:1 |0b1000|
X:56 |0b1000| 	Y:92 |0b0| 	Z:42 |0b1000| 	E0:1 |0b1000|
echo:busy: processing
X:56 |0b1000| 	Y:78 |0b1000| 	Z:76 |0b0| 	E0:1 |0b1000|
X:57 |0b1000| 	Y:80 |0b1000| 	Z:75 |0b0| 	E0:1 |0b1000|
X:55 |0b1000| 	Y:79 |0b1000| 	Z:70 |0b1000| 	E0:1 |0b1000|
X:59 |0b1000| 	Y:79 |0b1000| 	Z:69 |0b0| 	E0:1 |0b1000|
echo:busy: processing
X:58 |0b1000| 	Y:78 |0b1000| 	Z:65 |0b0| 	E0:1 |0b1000|
X:61 |0b1000| 	Y:45 |0b1000| 	Z:66 |0b0| 	E0:1 |0b1000|
X:58 |0b1000| 	Y:44 |0b1000| 	Z:65 |0b0| 	E0:1 |0b1000|
X:57 |0b1000| 	Y:48 |0b1000| 	Z:66 |0b0| 	E0:1 |0b1000|
X:56 |0b1000| 	Y:43 |0b1000| 	Z:60 |0b1000| 	E0:1 |0b1000|
X:58 |0b1000| 	Y:46 |0b1000| 	Z:63 |0b1000| 	E0:1 |0b1000|

Here I used the Home All command on the printer. To be honest, I do not know how to read these values  written above.
Can you help me with reading of these info?

Also, regarding heatsinks, I have ordered them, but I need to wait around 2 months for them to come from Aliexpress.

swilkens, 2018-10-18:
> Also, regarding heatsinks, I have ordered them, but I need to wait around 2 months for them to come from Aliexpress.

Given that people are reporting using heatsinks over 500 mAh, I'd say this seems likely to be the cause. The approach here would be to run the printer untill she skips a step, then call `M122` to see if OT was triggered.

> Strange thing for me is that on the same spot and same layer height, this missing steps occurs. They are not random.

I failed to notice this piece of information yesterday, can I ask how you are printing? Is this over USB or from an SD card? I'm thinking: failure at the same point means that either that point in the print is thusly taxing that the un-cooled stepper gives out, or you are printing from SD and the card is wonky, or you are printing over serial and the buffer runs out.

Yoonutz, 2018-10-18:
I have read over the other posts and some people said they went back to 1.1.6 Marlin version and that solved their issue. Other said they downgraded TMC2130 library. One even said he increased from 1/16 microsteps to 1/32 and that solved his issue. I will give those 3 things a try and see what happens. 
What I can do is change things here and there, untill heatsinks get here.