MarlinFirmware/Marlin #12127: [2.0.x] Last bugfix c90e26dfa6bf0a25e3094a94ea86931b844fb7f5 does not compile for me

Bergerac56, 2018-10-17:
Last bugfix (c90e26dfa6bf0a25e3094a94ea86931b844fb7f5) does not compile. It happened after 213e94bce2fd74b10014557bd6801b45193a08f8 which was compiling fine

I use specific pins for Case_Light. Does somethig have changed juste after release :  213e94bce2fd74b10014557bd6801b45193a08f8

In file included from Marlin\src\feature\../inc/../HAL/HAL_LPC1768/HAL.h:48:0,
from Marlin\src\feature\../inc/MarlinConfig.h:30,
from Marlin\src\feature\caselight.cpp:23:
Marlin\src\feature\caselight.cpp: In function 'void update_case_light()':
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/fastio.h:40:35: error: 'useable_hardware_PWM' was not declared in this scope
#define USEABLE_HARDWARE_PWM(pin) useable_hardware_PWM(pin)
^

Marlin\src\feature\caselight.cpp:73:9: note: in expansion of macro 'USEABLE_HARDWARE_PWM'
if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN))
^~~~~~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\feature\caselight.cpp.o] Error 1

Thanks

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2488171/Configuration.zip)

p3p, 2018-10-17:
Will be fixed soon, the Macro on lpc has no real purpose as it doesn't really need hardware PWM, just PWM, and all pins (upto 26 channels) are PWM capable (the macro should just return true), I need to get more random peripherals for my printer so I pick these issues up sooner, it was caused when I replaced the LPC platform.

Bergerac56, 2018-10-17:
Sure ;)

Bergerac56, 2018-10-18:
@p3p Hello Chris, Sorry for the delay. Not in the same timezone ;).
The fix solves the  compile problem. LEDs are working fine again with my config.

But the version 295422a0c9476b3ade1307b7778ae7fc306ab787 gives me a problem with SD card. I get a NO SD CARD.

p3p, 2018-10-18:
you may need to tweak the sbase pins files sd card options to suite your setup, it defaults to the onboard sd card now, and will need changed to `LPC_SD_LCD` if that is the one you are using.

Bergerac56, 2018-10-18:
Mmmm I did not notice those changes in sbase pin files. Changing to LPC_SD_LCD does not solve the access to the "LCD" sd card. It still does not detect it. I suppose it has to make with some changes to the pin used to detect the card. Looking and comparing the "old" sbase pin file and the "new" one shows more changes. I will have to investigate a bit ...

p3p, 2018-10-18:
The sd detect pin should only effect the auto-mounting, using M21 or selecting the option in the lcd should still give you access,

Bergerac56, 2018-10-18:
Unfortunately not... for the moment
M21 returns sdcard ok (with or without card)
M20 returns an empty list (with or without card)

p3p, 2018-10-18:
if M21 returns ok its finding an sdcard somewhere ^^ I presume its the onboard one that contains no gcode files.

Bergerac56, 2018-10-18:
You are right. Even if in an enclosure, I have access to the onboard sd card. When removed, I get a "card removed" on LCD and an init fail with m21

p3p, 2018-10-18:
okay so know we just need to know why the lcd option isn't switching it to the other card,

your settings should be as follows:
```
//#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD
#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive attached to the control board
```

Bergerac56, 2018-10-18:
Compile ongoing ;)

Bergerac56, 2018-10-18:
With a fresh sbase pin file, and no other modification that the #define LPC_SD_LCD, it does not work. M21 returns an init fail.

p3p, 2018-10-18:
Could you try slowing down the spi in te config, if that doesnt work change it to software spi(`#define LPC_SOFTWARE_SPI`).

p3p, 2018-10-18:
I just checked on my board and the signals are definitely going to the correct pins when setting it to `LPC_SD_LCD`, If you want more realtime support you can usually find me on the Marlin discord btw

Bergerac56, 2018-10-18:
Where I am:
When onboard card options are active, I can access the gcode files on the onboard card after a m21 or a init SD card on the LCD menu. 

Now... let's try to have the same behavior with the LCD SD card...

gloomyandy, 2018-10-18:
I think it would help if you posted the current state of your pins file and also confirmed what sort of cable you are using between the LCD and the SBase board. Are you using just a standard cable (with all pins just connected one to one) or are you using some sort of custom cable? It would also help if you described what configuration you would like to achieve. So what if anything do you want to use the onboard SD card for? Do you intend to have any sort of USB based file access, if so which SD card do you want to access?

Bergerac56, 2018-10-18:
@gloomyandy I use a custom cable. 4 lines are routed to j8 the way you can find back in a lot of posts or wiki. Everything was fine till the changes mentionned above. I use marlin for a long time now on different platforms(8 and 32 bits) and follow very closely each development. So I understand that thinks can change and I am more than ok with that. 

The MSK_SBASE pin file I use is the exact same file present in the last bugfix. The only change I make usually is to change the #define LED_PIN   to -1   in order to be able to use this pin for the RGB strip without facing (for me) a blicking red strip. (I use P1.18 /21/20 for RGB and P1.19 for Case_Light).

What I want to achieve? Just find back the behavior of the control panel I had before. This means a fully functionnal  graphic LCD (Which is OK for the moment), a fully functionnal encoder (OK too) and a usable SD card reader (Which does not work anymore even if it is the exact same harware configuration)

The changes made to the sd card access give now access to the onboard card. But, on my printer, I cannot access the lcd sd card anymore. 

I am trying various combinations between options to see if I can improve this situation. From now, the only stable result I achieved is to have access to the onboard sd card, but with a mandatory new step: initialyze the sd card (M21 or option on LCD menu).

Bergerac56, 2018-10-18:
I just tried. No luck.. for the moment. I cannot figure out yet which pins changed with the new way to access the SD card

gloomyandy, 2018-10-18:
See my edited comments above...

Bergerac56, 2018-10-18:
Basically, I always use the LCD SD card. The only use I do with the onboard card is to inject a new firmware. Before the modification mentionned above, this card was accessible through  USB which was comfortable. But this is not a must. I can access  the onbord card physically.

gloomyandy, 2018-10-18:
That's fine did you see my edited comment above that describes a configuration that I think should give you what you want? If you want to keep the USB access to the onboard card then just leave the USB_* defines as they are in the default pins file.

Bergerac56, 2018-10-18:
I used nearly the same changes (I did not kill the USB access)

Here my file. The configs files are in my first post

[pins_MKS_SBASE.zip](https://github.com/MarlinFirmware/Marlin/files/2493010/pins_MKS_SBASE.zip)

gloomyandy, 2018-10-18:
OK I got the config slightly wrong earlier. I've just built a version using the following and it works fine for me, using a custom cable the SD card on the LCD works just as expected (including detecting when the card is inserted). The onboard SD card is available via USB...
```
//#define USB_SD_DISABLED
#define USB_SD_ONBOARD        // Provide the onboard SD card to the host as a USB mass storage device

/*
 * There are a number of configurations available for the SBase SD card reader.
 * A custom cable can be used to allow access to the LCD based SD card.
 * A standard cable can be used for access to the LCD SD card (but no SD detect).
 * The onboard SD card can be used and optionally shared with a PC via USB.
 */

#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD
//#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive attached to the control board
```

Bergerac56, 2018-10-18:
@gloomyandy  It works !
Here I am a bit puzzled. Why do we have to "kill" both options ?:
//#define LPC_SD_LCD         
//#define LPC_SD_ONBOARD      

The only difference with tests I did is to comment #define LPC_SD_LCD   (which seems to me not logical...)

p3p, 2018-10-18:
I should have asked if you were using a custom cable, when you write things you think it's obvious but it never is to other people. 

You need to select only one connection method as they are independent, the LCD option means you are using the standard cable, you can't both be using the standard cable and a custom one. The config option naming and description probably need cleaned up to be more user friendly.

gloomyandy, 2018-10-18:
Yes you can only have one of those three options selected (we could probably add code to check that only one is defined). As @p3p explains the custom cable options means you will accessing the SD card in the LCD via a custom cable. Really that option will probably not be used much in the future as there is not really any need to use a custom cable anymore.

Bergerac56, 2018-10-19:
Thanks you both. I did not catch this subtility. And, yes, some more explanations in the comments could help.
Do I understand correctly? If I replace my custom cable by a standard one and choose the correct options, it will work? (Open the case of my printer and try asks me more time than to ask...). This will help for my other printers, not yet converted to 32 bits.

gloomyandy, 2018-10-19:
It will work but there unfortunately there are a couple of issues you will have to deal with because of the SBase hardware design....
1. The extra cable now attached to the shared SPI bus may mean that reading/writing to the onboard SD card does not work at the current speed.
2. On the SBase both SD cards share the same card inserted pin.

In more detail....
You may never see problem 1, it depends on your cables and the route they take on the printer. If you do see the problem you may find that moving the cables or shielding them fixes the problem. If this does not work then you/we can slow down the speed that the system accesses the onboard SD card (or you can simply not use the onboard SD card for any printing stuff). I have seen this problem on my setup and slowing down the access fixed the issue, but I don't want to make that the default as I have no idea how many others will see a problem.

The shared SD card inserted pin is a bit of a pain. You have a few options...
1. undefine the pin in the pins file and use the manual mount/unmount options from the menu or from remote software. Note that this is the default when using a standard cable.
2. Don't use the onboard SD card for anything other than flashing the firmware and leave the onboard card reader empty, then the shared pin will work fine with the LCD card reader.
3. Create a simpler custom cable with just the one custom pin for the card inserted pin.

Finally there is the option just to not use the LCD based SD card reader (the default option). I've found that with USB access to the onboard reader and the ability to see the same files from the printer this is the option that I use. I basically never remove the SD card from the printer, I just upload the files to the SD card over USB and then print them using the LCD. This option may not work for everyone (you may not have the computer you slice things on attached to your printer by USB), in which case I suggest that option 2 above may work well. 

NOTE: If you go for the onboard only option then a standard cable can still cause noise issues. But luckily there is a solution for this. The SBase hardware has a set of switches which lets you disconnect the shared SPI bus from the LCD connector.

I'm sure that over time when we have more feedback on things we will refine the various options and settings for this kind of thing. I'd be interested to understand your workflow and which setup you prefer and why.

Bergerac56, 2018-10-19:
@gloomyandy Thanks Andy for this explanation. It clarify a lot all the options. I love to experiment, but I am also "human". This means that when you have a working routine installed, you do not question it too often ;-)

For the moment, I have a sd card always in the onboard slot. I copy firmware through USB and this is the only access I do to this SD card. (Obviously I use also the USB connexion for direct piloting of the printer trough a host)

To print "remotely", I use 2 options. 
1> Extract the SD card from the LCD slot and copy the desired file on it. Printing occurs then from this card. 
2> Or, for testing purpose of other ways to put a screen on a printer (the printer I am speaking about is my "test" printer), I have also a TFT32 screen linked to serial port and "glued" on the printer. I can use the SD card attached to it too.

The idea of using only the onboard sd card could be a very good one as I do basically the same thing, but in a more cumbesome way (Transfer wia USB firmwares, but maniplate physically another sd card to print a 3D file). If I think about, it is a bit illogic from my side :-)

I will give a try to your method (All on the onboard card). To test that, I suppose I can keep my custom cable (So I do not have to reopen the case for the moment)

Olivier

Bergerac56, 2018-10-19:
@gloomyandy Andy, can you tell me how to slow down the access to the onboard card? Is it via the SPI options in configuration.h? (Or is this only for LCD card?)

When I go for all on the onboard sd card, I experience a very bad transfer rate. I already noticed that when  transfering a FW, but it was not really a concern for one file occasionnally. I spent a lot of time to route correctly all cables. (good old time of the temperature problem on REARM). But I have still to investigate the switches to desactivate on the SBASE.

Moreover, when I initialyse the card (M21 or LCD), I loose access to the card through USB. I still have acces via pronterface for ex. This means that if I want to copy a file to the onboard sd card, I have to do it before init card. Do you experience this also? (with only LPC_SD_ONBOARD and USB_SD_ONBOARD activated. No difference with custom cable activated or not)

gloomyandy, 2018-10-19:
Yes I know about the LCD menu item, I raised an issue about it some time ago to try to get feedback as to if it was deliberate choice. I didn't get much response..
https://github.com/MarlinFirmware/Marlin/issues/11809
I've just been changing the command used in my code.

As to the speed of the SD card when are you getting slow transfers to the the card and how slow are they? If it is via USB then remember that the SBase only supports USB "Full Speed" which is actually not very fast. I usually get about 350KB from Windows to the printer as displayed by windows. Which is not that fast but is still much faster than transferring the data by pronterface or whatever. I'm currently working on a new version of the USB code that uses DMA and this pushes that speed up to around 900KB. What size files are you transferring and how long do they take?

gloomyandy, 2018-10-19:
Copying a firmware file to the SD card via USB is pretty much instant for me. I'm not sure why your access is so slow. Is this something that has happened recently or has it always been slow for you? 

Have you tried different SD cards? Are you using any sort of hub between your PC and the printer? Can you try different  a USB port on your PC (in particular do you have one that is not a USB 3 port)? What OS are you running on your PC? You could try removing the cables that go to the LCD screen to see if that makes any difference, unplug them from the board.

Bergerac56, 2018-10-19:
Configuration:
Windows 10
No hubs
USB2 for one PC / USB3 for the other
No other software running at the same time
Very short USB cable (40cm)

Now let's play with the sd cards:

First try: Original card  was a Toshiba HC class 4- 8Gb (>10 sec for 207kb. Multiple files: let's forget it)
Second try: Unknown origin HC I class 10 - 32Gb (2 sec for 207kb/ average of 171kb/s for multiple files)
Third try : Samsung EVO class 10 - 32Gb ( Multiple files 430 Kb/s ! )
Fourth try : Sandisk Ultra HC I Class10 - 32Gb (Multiple files: av 177 kb/s)

Definitively, the type of sd card changes the game. I will stick to the Samsung EVO which is from far the best in my case. 

When I have some time, I will see if the cables play a role there too.

p3p, 2018-10-20:
> It takes ~10 to 25 sec before the SD card becomes accessible on the host (depending from the card). 

This is just how it is, The OS needs to read the FAT it takes time on slow media like this,

> If not ejected properly on windows, the printer has to be restarted in order to have access to the sd card through USB again.

Don't just pull out the sdcard when its mounted, it's a known limitation of the MSC stack, it doesn't detect the removal

> When used, the option Init SD Card / Change SD Card of the LCD menu becomes full garbage

Not sure whats causing this.

Could you close this issue as your original issue is solved, we can still have further discussion if needed.

gloomyandy, 2018-10-20:
Yes those good times are in line with what I typically see. With the DMA code things will be about 2-3 times faster. 

For a discussion on why it takes so long for the SD card to be ready see: https://github.com/MarlinFirmware/Marlin/issues/12128

Can you describe how you managed to get the SD card to be not available to Windows? I'm not aware of this at all. Normally in the worst case doing an SD Init followed by Change SD will restore access from Windows. I've not seen any problems when copying files over and then just using init SD from the LCD (with no eject under Windows, though really you should do this), to gain access to start a print, doing a change SD after the print completes restores Windows access (but you may have to wait for twice the start up time, because for some reason Windows reads the FAT twice! See the thread above).

Yes I see the same thing with the LCD screen when you select InitSD, I suspect it is a refresh problem, possibly not LPC1768 specific.

closed 2018-10-20 by Bergerac56

Bergerac56, 2018-10-26:
@gloomyandy Hello Andy,
I come back on the "Change SD Card" point. In ultralcd.cpp, the gcode refered by the "change sd card" option seems to be M21. On my printer, this leads to... no action and no change.

Here is the original piece of code.
    #if ENABLED(SDSUPPORT)
      if (card.cardOK) {
        if (card.isFileOpen()) {
          if (card.sdprinting)
            MENU_ITEM(function, MSG_PAUSE_PRINT, lcd_sdcard_pause);
          else
            MENU_ITEM(function, MSG_RESUME_PRINT, lcd_sdcard_resume);
          MENU_ITEM(function, MSG_STOP_PRINT, lcd_sdcard_stop);
        }
        else {
          MENU_ITEM(submenu, MSG_CARD_MENU, lcd_sdcard_menu);
          #if !PIN_EXISTS(SD_DETECT)
            MENU_ITEM(gcode, MSG_CNG_SDCARD, PSTR("M21"));  // SD-card changed by user
          #endif
        }
      }
      else {
        MENU_ITEM(submenu, MSG_NO_CARD, lcd_sdcard_menu);
        #if !PIN_EXISTS(SD_DETECT)
          MENU_ITEM(gcode, MSG_INIT_SDCARD, PSTR("M21")); // Manually initialize the SD-card via user interface
        #endif
      }
    #endif // SDSUPPORT

When I adapt the code this way (M21 replaced by M22), all works like expected:
         MENU_ITEM(submenu, MSG_CARD_MENU, lcd_sdcard_menu);
          #if !PIN_EXISTS(SD_DETECT)
            MENU_ITEM(gcode, MSG_CNG_SDCARD, PSTR("M22"));  // SD-card changed by user
          #endif
        }

Is it me or there is something there ?

Thanks

gloomyandy, 2018-10-26:
Personally I think that the GCode for change SD card should be M22, but when I raised it as a potential issue I did not really get much feedback. For now I just change the code in my repo and move on. At some point I'll probably put together a PR that makes the command used a configurable option. I suspect that things are as they are because many people just eject the SD card and then insert a different one and so want a quick way to mount the new card (even though they have not unmounted the old one), using M21 works better for this. But they could probably just use the option to refresh the SD card that is present in the print from SD card menu...

Bergerac56, 2018-10-26:
In fact I use it (change sd card) more to switch between onboard card and USB access (to the onboard card).

Yes, you convinced me to work only with the onboard card :)

This is also why I changed the label of the option by "Release SD card" and not "Change SD Card" 

This means that my routine is now:
When I want to print using the onboard card: I "Init" it
When I want to write to the onboard card from the host (through USB for gcodes or FW) I "release" it.

gloomyandy, 2018-10-26:
Yep that is pretty much what I do as well. Please feel free to add a comment to the issue I created about the  M21/M22 usage.

Bergerac56, 2018-10-26:
Done ;)