MarlinFirmware/Marlin #12128: [2.0.x, HAL_DUE, RFC] Revisting "sd_mmc_spi_mem" delay when inserting media

marcio-ao, 2018-10-17:
Earlier I had encountered a problem where I found the "sd_mmc_spi_mem.cpp" functionality was causing Marlin to hang for almost a minute after media is inserted (see https://github.com/MarlinFirmware/Marlin/issues/10755). I had gotten around the problem by modifying all the functions in "sd_mmc_spi_mem" to return CTRL_NO_PRESENT, but today I decided to investigate the problem further.

It appears as if the host OS (Debian Linux) is reading about ~3 MB of data from the device, before deciding it is a valid device. For reliability over a ribbon cable, we are running our SPI interface at about 200kHz. If you do the math, it works out to about 115 seconds to transfer that initial burst of data, which is approximately the delay I am seeing.

Interestingly, Marlin itself can initialize the device by reading only four blocks. I have attached a log that shows that Linux seems to do a pretty exhaustive linear search by reading about five thousand blocks on the device before mounting it.

[cutecom.log](https://github.com/MarlinFirmware/Marlin/files/2488410/cutecom.log)

The first four accesses before the "echo:SD card ok" are Marlin, the rest of the accesses are the host making accesses via the "sd_mmc_spi_mem" interface.

I am curious whether anyone has seen this problem on other OSes? It seems like having a feature in Marlin where the host OS can essentially do a denial-of-service attack on the printer is a bit worrisome, especially when that feature is enabled by default and can't be turned off?

I guess I am suggesting this media access feature ought to a user configurable option. Thoughts?

ejtagle, 2018-10-17:
@marcio-ao : Yes, i saw that. On both Windows and Linux, the OS reads the whole FAT to cache it as soon as you insert media. There is no workaround,
That is why i had to implement the SPI sw communication in assembler: To raise the read speed as much as possible.
The other choice would be to stop using MSD and start using MTP as the protocol.
That would allow to share the SD between the Marlin and the host at the same time.
I started the implementation of a MTP host (there is none available as open source for embedded targets), and the spec is not as complex, but i still have to finish it..
MTP works over the CDC USB class, so it could be shared between all 32bit boards...

ejtagle, 2018-10-17:
The other thing is that if you use an SPI communication at 200khz, there is no point in using the SD card as MSD. It would be extremely slow ...

marcio-ao, 2018-10-17:
> The other thing is that if you use an SPI communication at 200khz, there is no point in using the SD card as MSD. It would be extremely slow ...

Yes. I agree. We've pretty much decided that we must turn this feature off in our printers -- the only thing I am questioning is whether this should be enabled by default in Marlin.

> The only thing i had to do was to switch to a shielded ribbon cable. And it works with no issues at all

I would suspect very few printers are actually build to that standard. A lot of printers, such as the ones that use a RepRapDiscount display, use unshielded ribbon cables and I wonder whether they would be able to support this feature. It seems like there is a big disconnect between the speed in which host OSes expect MSD to operate (>10 MHz) and what most printers out there are built to operate at (100kHz-500kHz).

gloomyandy, 2018-10-17:
Yep I saw exactly the same thing with Windows when working on the LPC1768 SPI and USB code. As mentioned above Windows reads all of the FAT into memory when it "mounts" the drive. Even worse for some reason with Windows 10 if you mount/unmount/mount the same card it seems to read the FAT twice on the second mount! 

On the LPC1768 we do not use the Marlin SD/SPI code when accessing the SD card from the USB stack but instead use code from chanfs. This code runs the (hardware) SPI bus at 25MHz and works fine with the onboard SD card reader. However on the MKS SBase board the same SPI bus is also used to talk to the external SD card and if a standard cable is used, just having the cable connected seems to cause problems and I had to reduce the speed to 12MHz  to get access to the onboard SD card to work (The MKS board contains a set of header switches that allows you to disconnect the external SD card, perhaps for this reason). On the LPC1768 the PC can not lock out the printer from access to the SD card, the printer access has priority (and will cause the PC access to fail).

I'm a little surprised you have to run the SPI bus at such a low speed. On the SBase I can happily access the external SD reader (via the shared hardware SPI bus) at the maximum "Marlin speed 0" which seems to be 8.3MHz. This is with an un-shielded ribbon cable that is about 40cm long.

On My setup with a 16Gb card Windows was taking about  10 seconds to mount the onboard SD card. I've recently reworked the USB MSD code to use DMA and 2 4KB buffers (the LPC1768 has a 16KB memory area set aside for USB DMA use), with this configuration the mount time drops to about 5 seconds (these times from memory, but that sort of ballpark).

ejtagle, 2018-10-17:
Well, i am using a reprapdiscount display. 
I had to shield the ribbon cable. But, i know a lot of people that are having issues even with AVR and reprapdiscount displays. Seems the shielding is mandatory if the ribbon cable is near power cabling.

I still think MTP is the proper answer to all this problems

marcio-ao, 2018-10-17:
> I'm a little surprised you have to run the SPI bus at such a low speed.

Well, there's more to the story that I haven't mentioned which may explain why we are needing to drop the data rate so low. We are actually daisy chaining three boards together using two unshielded ribbon cables. There are a total of 3 SPI devices, the first being a SPI flash on the Archim board itself. So it looks like this:

```
    Archim Board                   Color LCD Board                      USB Host Board
     SPI Flash        <------->   FTDI 810 Controller    <------->         MAX3421E
```

Given that all this is done with unshielded ribbon cables, it's probably not surprising that I am having to drop the data rate this low. We have run the SPI interface as high as 1 MHz, but at that rate problems crop up occasionally, and 200 kHz seems to be a good speed for a rock solid connection.

It seems that without upgrading to shielded cables, this is the best we can do and using the MSD functionality is entirely out of the question.

gloomyandy, 2018-10-17:
I agree that MTP would be a better alternative for USB access. Does it just work out of the box on most systems? The Wikipedia page seems to indicate that some systems (Windows) may need specific vendor/product IDs, but I've no idea if that is (still) the case. 

I suspect that a shielded cable would have allowed me to run the external SD card at 25MHz but I didn't try it. I'm surprised that @marcio-ao had to drop the speed to 200KHz to get it to work (ahh I've just seen his post!).

ejtagle, 2018-10-17:
@marcio-ao , @gloomyandy : If you dont want to use shielded cable, there is anothe choice: Do you remember 80wire IDE ribbon cables ? -- The idea is that using proper interleaving you can also reach those 25mhz.
The ribbon cable should have a ground (or vcc) cable between all data cables. It tends to act as shield and also improves quite a bit crosstalk.
The other thing that could be added is proper termination to the lines. A 100ohm resistor at the end of the cable is a good idea

marcio-ao, 2018-10-17:
> As mentioned above Windows reads all of the FAT into memory when it "mounts" the drive ... Even worse for some reason with Windows 10 if you mount/unmount/mount the same card it seems to read the FAT twice on the second mount!

This must be the reason that even though our PCs hardware gets faster every year, it feels as sluggish as it always had! That's crazy, reading all that data upfront and even doing it twice!

> The ribbon cable should have a ground (or vcc) cable between all data cables. It tends to act as shield and also improves quite a bit crosstalk.

Early on in our design, we made the choice to maintain compatibility with the connector for the RepRapDiscount display, since our controller boards already have that style of connector. This means that all 10 lines on the ribbon cable are used for signaling. Perhaps we could squeeze a little more speed out of the cable by interleaving the chip select lines between the signal lines, but it's probably too late for us to rev our design that way.

marcio-ao, 2018-10-17:
This is what our cable looks like right now:

1. MISO
2. SCLK
3. Display Enable
4. USB CS
5. Display CS
6. MOSI
7. USB Interrupt
8. RESET
9. GND
10. VCC

In hindsight, it probably would have worked much better if MISO wasn't right next to SCLK :(

marcio-ao, 2018-10-17:
> You can improvise a shielded cable easily: Get some aluminium foil (the one that is used for cooking works), and wrap the cable.

@ejtagle: Heh. I don't think our customers would like us to sell printers like that :D

p3p, 2018-10-17:
Surely the overhead in cost for shielded cabling has to be worth the reduction in support and increase in performance? Even if the design needs tweaked slightly to get the shielding earthed in the connectors.

ejtagle, 2018-10-17:
> This is what our cable looks like right now:
> 
> 1. MISO
> 2. SCLK
> 3. Display Enable
> 4. USB CS
> 5. Display CS
> 6. MOSI
> 7. USB Interrupt
> 8. RESET
> 9. GND
> 10. VCC
> 
> In hindsight, it probably would have worked much better if MISO wasn't right next to SCLK :(

Switching VCC and SCLK, MOSI and RESET, Display enable and USB interrupt would probably do the trick (probably the first is the most important). And proper termination of lines.
If you inspect reprapdiscount wiring, you will notice that MOSI/MISO/CLK are properly interleaved ;)

marcio-ao, 2018-10-17:
> Surely the overhead in cost for shielded cabling has to be worth the reduction in support and increase in performance? Even if the design needs tweaked slightly to get the shielding earthed in the connectors.

@p3p: Unfortunately, it's not my call at this point. I don't think MSD is a strong enough use case to persuade anyone to make changes.

The only real downside I have seen to the 200kHz signalling rate is that the USB device gets really slow when you have about 100 files on the USB drive. But this isn't something that can be fixed by doubling or tripling the signaling rate, however. The reason for that is that when displaying a file list, for each file it displays, it rereads the file list up to that file. So the amount of time to display files from a card having n files actually increases at roughly n*n.

> Switching VCC and SCLK, MOSI and RESET, Display enable and USB interrupt would probably do the trick (probably the first is the most important).

@ejtagle: I wish we had considered this months before. We would have to redesign two boards to make this change now!

marcio-ao, 2018-10-17:
@ejtagle: I am personally in favor of making these changes, but if it were up to me I don't think we'd ever get a product out the door -- there is *always* some way the design can be improved!

I'll certainly bring these suggestions up, but at this point I am not optimistic.

ejtagle, 2018-10-17:
For the next model, or perhaps when running out of PCBs this mod should be done. There is no extra cost, but certainly improvements

marcio-ao, 2018-10-17:
@ejtagle: I agree. We need to start a "lesson's learned" notebook for our next iteration :)

marcio-ao, 2018-10-17:
I think I found a bug in the "sd_mmc_spi_mem.cpp" code:

```
Ctrl_status sd_mmc_spi_read_capacity(uint32_t *nb_sector) {
  if (!IS_SD_INSERTED || IS_SD_PRINTING || IS_SD_FILE_OPEN || !card.cardOK)
    return CTRL_NO_PRESENT;
  *nb_sector = card.getSd2Card().cardSize();
  return CTRL_GOOD;
}
```

Should be:

```
/* NOTE: This function is defined as returning the address of the last block
 * in the card, which is cardSize() - 1 */
Ctrl_status sd_mmc_spi_read_capacity(uint32_t *nb_sector) {
  if (!IS_SD_INSERTED || IS_SD_PRINTING || IS_SD_FILE_OPEN || !card.cardOK)
    return CTRL_NO_PRESENT;
  *nb_sector = card.getSd2Card().cardSize() - 1;
  return CTRL_GOOD;
}
```

This is because `sd_mmc_spi_read_capacity` [is defined](http://asf.atmel.com/docs/latest/uc3b/html/group__group__avr32__components__memory__sd__mmc__sd__mmc__spi__mem.html#gaea77d33e127be3f7439fc3718794b61e) as returning "the address of the last valid sector.", which is one less than the capacity.

Prior to this change, I determined the host OS would try reading a sector which was one past the end of the card. Fixing the function keeps the OS from making that incorrect block request.

I was hoping that fixing this would make the card initialization faster (because maybe the OS was failing to find the FAT), but it did not help much at all.

ejtagle, 2018-10-18:
Unfortunately, there is no way to convince the OS (windows or linux) to avoid prefetching the whole FAT.
The only hope is to implement MTP (just because then the code responsible for reading the FAT will be in Marlin, not in the OS)
MTP basically is some packet s with orders like:
-read directory
-delete file
-move file
-read file
-write file

So all access to the media is done by Marlin, not the OS
The only problem is that MTP requires to create a "handle" for each object (file or directory) present on the card, and that requires memory, unless i find a way to perform a mapping mathematically (for example, using the start cluster number as the handle)...
But that is not as easy as it sounds.

closed 2019-02-14 by marcio-ao