MarlinFirmware/Marlin #12135: Missing implementation of "Tone" on LPC1678

andyman198, 2018-10-18:
### Description

Would like Tone() to be implemented on the LPC HAL

I have looked and am not sure there is a timer free for this, or if there is just not a timer implemented that can be used. I have also tried copying some of the DUE HAL Tone code to see if I can make it work and have been unsuccessful.

gloomyandy, 2018-10-18:
What display/controller do you intend to use this with?

andyman198, 2018-10-18:
AZTEEG X5 GT with a VIKI2

gloomyandy, 2018-10-18:
Are you sure that the VIKI2 has a speaker? The files here seem to say it has a buzzer:
http://panucattdevices.freshdesk.com/support/solutions/articles/1000158562-viki-lcd-2-0-support-files

andyman198, 2018-10-18:
happy to be proved wrong, but I believe it to be an SMT piezo. I have applied 3v and 5v across the buzzer and I don't get a tone, but as I connect and disconnect a voltage, or manually oscillate I do I get an audible noise. I had thought that the tone function produces a wave to do the oscillations for a piezo.

gloomyandy, 2018-10-18:
Hard to tell without more information or the actual device. Smoothieware seems to drive it with a pulse train. Just out of interest did you try connecting the pin to ground rather than 3v/5v? Anyway you could try driving the pin using PWM.

p3p, 2018-10-18:
@andyman198 I just added an implementation of tone to the lpc176x framework, running `pio update` should download version 0.0.3, I don't have speaker to test Marlin with but it made the right waveform.

andyman198, 2018-10-19:
Thank you! I will test and report back as soon as I can!

andyman198, 2018-10-20:
so isn't this odd... .first off … I can get tones of different frequencies which is awesome, worked straight out of the gate with the Azteeg and the Viki2 !!! however, not all the tones seem to be played properly, take for example the gcode below for 5 tones separated by motor movement, randomly not all the tones are played. each time I execute the code it could be a different set of tones that could be played, maybe 3 or 4  out of 5 will work fine, but the motors move as expected. Definitely a step in the right direction though, thank you p3p! im not sure why the tones don't all play though.

M300 S300 P500
g1 x0 f3000
M300 S350 P500
g1 x100 f3000
M300 S400 P500
g1 x0 f3000
M300 S460 P500
g1 x100 f3000
M300 S490 P500
g1 x50 f3000

p3p, 2018-10-20:
Read the Arduino Tone api slightly wrong and didn't allow overlapping tone calls on the same pin, but that is only suppose to happen if the call is for another pin so it doesn't interrupt the previous tone call. I put out a new version that hopefully fixes it.

andyman198, 2018-10-20:
I updated to 0.0.4, but it still seems unpredictable. Its odd, it will play the first few but then it goes silent and wont play any more. No tones will then work until a full reset of the board is done. 

I thought it might be serial related issues because I was having plenty before the LPC reorg earlier in the week (which fixed all the serial issues)  all but it doesn't seem to be -  I tried running the gcode from the SD card and it has the same problem. 

Thank you for all your help with this p3p

p3p, 2018-10-20:
I'l look into it, my tests don't show an issue using the functions directly (with the tone function), can play as many sounds as I want, but it may be interacting with Marlin badly.

forkoz, 2018-10-20:
Glad to see this working, no more beeping when I move through the menus.

p3p, 2018-10-20:
ok, so erm.. 0.0.5 anybody?

andyman198, 2018-10-20:
YOU DID IT!!! It works perfectly!!! THANK YOU!

In the process of testing 0.0.5 I found a library with 10,000 converted phone ringtones as gcode files. Tested your revision with a bit of Bryan Adams :)

you can find them here if you want to test with some tunes 
http://doc.3dmodularsystems.com/plays-a-music-at-the-end-of-a-print/

p3p, 2018-10-20:
It's a good job I started the versioning at 0.0.1 ;) glad to have that working for you, one more feature down for parity with the AVR version.

If its working properly after your testing make sure you close this issue.

andyman198, 2018-10-20:
I will close it now! I lm not sure there much more to go on party, I know you saw the neopixel issue. It’s really close. Thanks again.

closed 2018-10-20 by andyman198