MarlinFirmware/Marlin #12144: [bugfix-2.0] incompatible TMCStepper v0.2.2?

ar0707, 2018-10-18:
`WARNING: library TMCStepper claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (sam) architecture(s).`

Downloaded  today the lasted Marlin 2.0, after downloading the new TMCStepper library, this warning appear. 
After uploading the .bin to the Arduino Due with Smart Ramps, the Display is blank and I'm not able to controll the printer.

Flashing an old Marlin which uses the TMC2130 Library, everthing works.  

Thanks

thinkyhead, 2018-10-26:
Recommend you stick with the old code until @teemuatlut has completed `TMCStepper`.

boelle, 2019-02-21:
@ar0707 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

stormy-gh, 2019-07-09:
I think the problem is in stepper_indirection.cpp in "restore_stepper_drivers()".

Error condition:
In Configuration.h enable:
#define X_DRIVER_TYPE  TMC2130
#define PSU_CONTROL
#define AUTO_POWER_CONTROL
In there is no connection with TCM2130 Marlin hangs.

Problem is that in restore_stepper_drivers() method stepperX.push() is called without any check if communication with TMC2130 is working.

thinkyhead, 2019-07-11:
> restore_stepper_drivers

@teemuatlut — What would be the proper thing to check before we try to talk to the drivers? And, if we can't talk to the drivers at boot time, as I recall we do refresh them whenever the power is enabled by `M80` (or the auto-power behavior), so … we just have to not go into an infinite loop if the steppers are un-powered, whether at boot or any other time.

stormy-gh, 2019-07-11:
Because it issue is closed i made separate issue for this topic: https://github.com/MarlinFirmware/Marlin/issues/14558