MarlinFirmware/Marlin #12149: Hotend Temp Instability On Latest 2.0 Version

556duckvader, 2018-10-19:
I have recently downloaded the latest version of 2.0 running a RAMPS/ReArm and I have been having issues with the Hotend temp not being stable.
[image]

AnHardt, 2018-10-19:
How much is that wave peak to peak? 2°C? Else locking beautiful.
You could try to reduce Kp.

swilkens, 2018-10-20:
Is there actually a notable change between this and the previous version? What was the temperature fluctuation on your previous install? Can you go back to that previous version and check what the readings are?

Based on what we can see here, your hotend temp just swung upwards which is listed as 205.7 C, 0.7 C over the setpoint. That pretty good.

closed 2018-10-26 by thinkyhead